open-sdr/openwifi #192: There is a problem restarting the circuit board to load a new FPGA bitstream after updating the FPGA

8663, 2022-07-27:
After I update the FPGA according to the routine, regenerate the boot.bin file, and re mount it to the partition, the restart development board cannot start.The following is after restart。
After ---[ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000000b
, there was no response
[image]
[image]

The operating system we use is Ubuntu16.04，vivado is 2018.3 ,board is antsdr

JiaoXianjun, 2022-07-27:
You need to tell us your each step operation/command and their outputs.

8663, 2022-07-27:
[image]
In the figure above, we replace the.Hdf file and.Bit file with the two files we generated. 
Then execute：
$OPENWIFI_DIR/user_space/boot_bin_gen.sh $OPENWIFI_DIR $XILINX_DIR $BOARD_NAME
scp $OPENWIFI_DIR/kernel_boot/boards/$BOARD_NAME/output_boot_bin/BOOT.BIN root@192.168.10.122:
On board：
mount /dev/mmcblk0p1 /mnt
cp ~/BOOT.BIN /mnt
umount /mnt
reboot

JiaoXianjun, 2022-07-27:
Can you use our original openwifi-hw (no any modifications) to generate hdf and bit, then put it onboard to test? Is it ok or it has problem?

Which sd card image are you using?

8663, 2022-07-27:
[image]
We generate SD by directly burning the image file.

[image]
[image]
[image]
This is the version of openwifi HW we use. There is no problem updating it. However, there is a problem with mounting and restarting after replacing.Hdf and.Bit files to generate boot.bin. 
Is there a problem with my FPGA update operation? How to generate the boot.bin file after we modify the FPGA part

JiaoXianjun, 2022-07-27:
We never use windows to burn the image and we know there are always some issues related to this sd card image burned by windows by some users.  we suggest you use Linux and do not do any change to our code to go through the whole flow to make sure you can do all operations successfully.

8663, 2022-07-28:
We download openwifi-hw, generate the BOOT.bin file from the bit file and hdf file, and then update the FPGA. It can run normally. /wgh.sh file.
[image]
However, when we update the FPGA by recompiling the bit file and hdf file generated by openwifi-hw（We operate with reference to build FPGA in openwifi HW without any modification）, the ad9361 device cannot be recognized when running./wgh.sh. How can we solve this problem.

[image]

JiaoXianjun, 2022-07-28:
Good. Please paste all your operations and outputs here. Including screenshot of Vivado completion, all your commands, etc.

8663, 2022-07-28:
Download command：git clone --branch antsdr https://github.com/open-sdr/openwifi-hw.git

Update FPGA:
$OPENWIFI_DIR/user_space/boot_bin_gen.sh $OPENWIFI_DIR $XILINX_DIR $BOARD_NAME
scp $OPENWIFI_DIR/kernel_boot/boards/$BOARD_NAME/output_boot_bin/BOOT.BIN root@192.168.10.122:
mount /dev/mmcblk0p1 /mnt
cp ~/BOOT.BIN /mnt
umount /mnt
reboot

This is the output result after generating the boot.bin file

[image]
[image]
[image]
[image]
[image]

8663, 2022-07-28:
[image]

JiaoXianjun, 2022-07-28:
You put lots of commands from our readme, not the command you input/run in your computer environment. And you do not put the command outputs.  So, hard to tell the problem. 

Can you put here step by step of “your own” operations in your computer? Not just repeat our readme. And also the outputs. For instances what is the output of your boot_bin_gen.sh ?

8663, 2022-07-29:
Install the evaluation license of Xilinx Viterbi Decoder into Vivado.

export XILINX_DIR=/opt/vivado2018.3
 ./prepare_adi_lib.sh $XILINX_DIR
[image]

 export BOARD_NAME=antsdr
 ./prepare_adi_board_ip.sh $XILINX_DIR $BOARD_NAME

[image]
[image]

Launch Vivado:
 cd /home/zzn/getwifi/openwifi/openwifi-hw/boards/$BOARD_NAME/
source $XILINX_DIR/Vivado/2018.3/settings64.sh
vivado

8663, 2022-07-29:
[image]

In Vivado:
source ./openwifi.tcl
Open Block Design
Tools --> Report --> Report IP Status
Generate Bitstream
(Will take a while)
File --> Export --> Export Hardware... --> Include bitstream --> OK
File --> Launch SDK --> OK, then close SDK

[image]

8663, 2022-07-29:
We haven't made any changes to the project. The generated .Bit and .Hdf update boot.bin will cause the problem of unable to find the ad9361 device. Updating boot.bin with .Bit and .Hdf provided by you can work normally.

8663, 2022-07-29:
[image]
Excuse me, where can we find this number

8663, 2022-07-29:
[image]
How should this problem be solved

JiaoXianjun, 2022-07-29:
You keep asking how to solve your problem, but never show a full logs of operations and outputs. Without knowing your operations and outputs we do not know why. 

And you are not following our latest openwifi-hw readme, you are using some old versions. This could also be a problem.

8663, 2022-08-01:
root@analog:~# dmesg
Booting Linux on physical CPU 0x0
Linux version 4.14.0-gb6e3799-dirty (microphase@microphase-dev) (gcc version 7.3.1 20180314 (Linaro GCC 7.3-2018.04-rc3)) #1 SMP PREEMPT Thu Aug 12 12:24:19 CST 2021
CPU: ARMv7 Processor [413fc090] revision 0 (ARMv7), cr=18c5387d
CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
OF: fdt: Machine model: Analog Devices ADRV9364-Z7020 (Z7020/AD9364)
OF: fdt: earlycon: stdout-path /amba@0/uart@E0001000 not found
Memory policy: Data cache writealloc
cma: Reserved 128 MiB at 0x38000000
On node 0 totalpages: 262144
free_area_init_node: node 0, pgdat c0c54340, node_mem_map ef7f7000
  Normal zone: 1536 pages used for memmap
  Normal zone: 0 pages reserved
  Normal zone: 196608 pages, LIFO batch:31
  HighMem zone: 65536 pages, LIFO batch:15
random: fast init done
percpu: Embedded 16 pages/cpu @ef7c8000 s33036 r8192 d24308 u65536
pcpu-alloc: s33036 r8192 d24308 u65536 alloc=16*4096
pcpu-alloc: [0] 0 [0] 1
Built 1 zonelists, mobility grouping on.  Total pages: 260608
Kernel command line: console=ttyPS0,115200 root=/dev/mmcblk0p2 rw earlycon rootfstype=ext4 rootwait cpuidle.off=1
PID hash table entries: 4096 (order: 2, 16384 bytes)
Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
Memory: 895988K/1048576K available (7168K kernel code, 340K rwdata, 2376K rodata, 1024K init, 160K bss, 21516K reserved, 131072K cma-reserved, 131072K highmem)
Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xf0800000 - 0xff800000   ( 240 MB)
    lowmem  : 0xc0000000 - 0xf0000000   ( 768 MB)
    pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
    modules : 0xbf000000 - 0xbfe00000   (  14 MB)
      .text : 0xc0008000 - 0xc0800000   (8160 kB)
      .init : 0xc0b00000 - 0xc0c00000   (1024 kB)
      .data : 0xc0c00000 - 0xc0c551e0   ( 341 kB)
       .bss : 0xc0c5ab38 - 0xc0c82e84   ( 161 kB)
Preemptible hierarchical RCU implementation.
        RCU restricting CPUs from NR_CPUS=4 to nr_cpu_ids=2.
        Tasks RCU enabled.
RCU: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=2
NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
efuse mapped to f0802000
slcr mapped to f0804000
L2C: platform modifies aux control register: 0x72360000 -> 0x72760000
L2C: DT/platform modifies aux control register: 0x72360000 -> 0x72760000
L2C-310 erratum 769419 enabled
L2C-310 enabling early BRESP for Cortex-A9
L2C-310 full line of zeros enabled for Cortex-A9
L2C-310 ID prefetch enabled, offset 1 lines
L2C-310 dynamic clock gating enabled, standby mode enabled
L2C-310 cache controller enabled, 8 ways, 512 kB
L2C-310: CACHE_ID 0x410000c8, AUX_CTRL 0x76760001
zynq_clock_init: clkc starts at f0804100
Zynq clock init
sched_clock: 64 bits at 333MHz, resolution 3ns, wraps every 4398046511103ns
clocksource: arm_global_timer: mask: 0xffffffffffffffff max_cycles: 0x4ce07af025, max_idle_ns: 440795209040 ns
Switching to timer-based delay loop, resolution 3ns
clocksource: ttc_clocksource: mask: 0xffff max_cycles: 0xffff, max_idle_ns: 537538477 ns
timer #0 at f080c000, irq=17
Console: colour dummy device 80x30
Calibrating delay loop (skipped), value calculated using timer frequency.. 666.66 BogoMIPS (lpj=3333333)
pid_max: default: 32768 minimum: 301
Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
CPU: Testing write buffer coherency: ok
CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
Setting up static identity map for 0x100000 - 0x100060
Hierarchical SRCU implementation.
smp: Bringing up secondary CPUs ...
CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[... 171 lines cut ...]
init: isc-dhcp-server main process (1982) terminated with status 1
init: isc-dhcp-server main process ended, respawning
init: isc-dhcp-server main process (2000) terminated with status 1
init: isc-dhcp-server main process ended, respawning
init: isc-dhcp-server main process (2013) terminated with status 1
init: isc-dhcp-server main process ended, respawning
init: isc-dhcp-server main process (2039) terminated with status 1
init: isc-dhcp-server main process ended, respawning
init: isc-dhcp-server main process (2061) terminated with status 1
init: isc-dhcp-server main process ended, respawning
init: isc-dhcp-server main process (2069) terminated with status 1
init: isc-dhcp-server main process ended, respawning
init: isc-dhcp-server main process (2077) terminated with status 1
init: isc-dhcp-server main process ended, respawning
init: isc-dhcp-server main process (2085) terminated with status 1
init: isc-dhcp-server main process ended, respawning
init: isc-dhcp-server main process (2093) terminated with status 1
init: isc-dhcp-server main process ended, respawning
init: isc-dhcp-server main process (2101) terminated with status 1
init: isc-dhcp-server main process ended, respawning
init: isc-dhcp-server main process (2109) terminated with status 1
init: isc-dhcp-server main process ended, respawning
init: isc-dhcp-server main process (2117) terminated with status 1
init: isc-dhcp-server respawning too fast, stopped
[image]

closed 2022-11-02 by 8663