open-sdr/openwifi #194: Hello, may I ask whether Openwifi has time slot division in FPGA

zhuzhineng, 2022-07-29:
Hello, may I ask whether Openwifi has time slot division in FPGA

JiaoXianjun, 2022-07-29:
We explains time slot/slicing here: https://github.com/open-sdr/openwifi/blob/master/doc/README.md#sdrctl-command

you can also find lots of tdma work in the openwifi publications:
https://github.com/open-sdr/openwifi/blob/master/doc/publications.md

zhuzhineng, 2022-08-01:
This problem will not occur when using the precompiled image you provided, but will occur when compiling the source code using Vivado 2018.3

Using the bitstream file provided in the precompiled image provided by you to regenerate the boot.bin file will not cause this problem. Is there a solution
[image]

JiaoXianjun, 2022-08-01:
This is a different issue. 

Please open a new issue and describe the openwifi and ooenwifi-hw revision/commit-hash when you have this issue. Also other info please, such as board, Linux image, etc.

zhuzhineng, 2022-08-01:
At that time, we replaced your original bitstream and DHF files with our own generated bitstream and DHF files, and then generated boot.bin files, and then performed the basic operations on GitHub. The following problems were encountered, is there any way to solve them?
[image]
[image]
[image]

zhuzhineng, 2022-08-01:
How to use our own bitstream and HDF files to generate boot.bin files

zhuzhineng, 2022-08-02:
When we modified the FPGA project, we generated new bitstream and HDF files, how to apply to openWiFi project

peking428, 2023-03-10:
make BOOT.bin ,include bitstream  zimage  uboot  rootfs .  have some step .