open-sdr/openwifi #196: Modify the operating frequency before connecting

reuncc, 2022-08-02:
I have two antsdrs with openwifi installed.
    I want them to work at 1444Mhz.
    If the sdrctl instruction is used, it can only be set after the device is connected, so if the link is broken after connection, it cannot be reconnected.
     I have downloaded the source code of "iwconfig wpa_supplicant hostapd". I want to know whether modifying these source codes by single pass can modify the frequency before connecting and keep the device working normally at the frequency I specify 1444mhz. If not, what else needs to be modified to meet the requirements.
      Or what other ideas can achieve such a goal.

JiaoXianjun, 2022-08-02:
“ if the link is broken after connection, it cannot be reconnected” can you describe the steps to reproduce the issue? All commands you run after the two boards boot

lydenlau, 2022-08-10:
> ```
> I have two antsdrs with openwifi installed.
> I want them to work at 1444Mhz.
> If the sdrctl instruction is used, it can only be set after the device is connected, so if the link is broken after [...]

hello reuncc，I have the same needs as you，did you run it at 1444Mhz successfully?