open-sdr/openwifi #197: Linux canoot boot with SD Card

enriqeat, 2022-08-02:
Hi. I created the Image from scratch on the SD Card of my Zedboard. However, Linux is not booting, The image attached shows some information gathered using serial communication. 

Do I need to manually create this missing xillydemo.bit file? or am I missing some part of the process? How can I solve this ?

Thanks.
[image]

JiaoXianjun, 2022-08-02:
I do not know, unless you offer info about all operations and outputs you have when you create your sd card image.

enriqeat, 2022-08-05:
I attach the output from the terminal, and an image of what I got afterwards on the SD card. This is the starting point that leads to my original question.

[Terminal_sd_card_output.txt](https://github.com/open-sdr/openwifi/files/9265901/Terminal_sd_card_output.txt)

[image]

Afterwards, I added to the BOOT directory, a xillydemo.bit file that I found on another repo. And I got this result with the serial communication.

 
[serial_comm_output_with_xillydemobit.txt](https://github.com/open-sdr/openwifi/files/9265925/serial_comm_output_with_xillydemobit.txt)

JiaoXianjun, 2023-04-28:
Please try with our latest scripts.