open-sdr/openwifi #201: ./wgd.sh issue

chubohao, 2022-08-14:
Hello Dr.Jiao,

My board is MicroPhase ANTSDR AD9363, I followed the steps below, but had problems executing the . /wgd.sh command, I encountered a problem.

1. Downloaded mirrors in the table.
2. Using the image burning software (Windows), burn the image to the SD card and power on the boot.
3. Execute cd openwifi, then execute . /wgd.sh, and then encountered the following error message.

[image]

Best wishes and looking forward to your reply.

Bohao

black-pigeon, 2022-08-15:
I think you can refer to this https://github.com/open-sdr/openwifi#build-openwifi-linux-img-from-scratch, to build the image from scratch. I have tried this and it works for me.
you can uncomment some instructions to speed up the build process, for example you can uncomment the scripts related to linux-64, which is for mpsoc device.

JiaoXianjun, 2022-09-29:
"Using the image burning software (Windows), burn the image to the SD card."

As far as I know, lots of user have issue with Windows (even for the SD card burning). I suggest you use Linux (such as Ubuntu 18 or 20 LTS).

hitanvil, 2022-11-16:
I and several other guys use AndSDR also met this problem. Actually, I did not build it but just downloaded the image from https://users.ugent.be/~xjiao/openwifi-1.3.1-wilsele-1-32bit.img.xz. Restore it to SD card. But wgd.sh fails.

I digged into the flow. The problem happens at load_fpga_img.sh line 62, it fails and exit to wgd.sh, CLI report that wgd line 142 segment fault.

I tried to comment out line 62. Then load_fpga_img.sh runs to line 74 and loop forever as it cannot find device  /sys/bus/platform/drivers/cf_axi_adc/79020000.cf-ad9361-lpc.

I had tried to build with latest master. It has no problem to support zc702+fmcomms4, but it still fails for AntSDR.

hitanvil, 2022-11-16:
Asked AntSDR support, they said that they can get good build by building from scratch. I will try it tomorrow.

hitanvil, 2022-11-17:
Create a SD card with the image from https://users.ugent.be/~xjiao/openwifi-1.3.1-wilsele-1-32bit.img.xz. Then generate bitstream and driver with latest code. Update SD card with methon in section Reload driver and FPGA(https://github.com/open-sdr/openwifi/blob/master/doc/app_notes/drv_fpga_dynamic_loading.md). Then run wgd.sh.It repeats below information for ever:
+ '[' '!' -d /sys/bus/spi/drivers/ad9361/spi0.0 ']'
+ echo 'Waiting for /sys/bus/spi/drivers/ad9361/spi0.0'
Waiting for /sys/bus/spi/drivers/ad9361/spi0.0
+ sleep 0.2

hitanvil, 2022-11-19:
Today I used Linux to re-burn SD card again. Different from previous operation, I did not touch computer during burning. Now the image from https://users.ugent.be/~xjiao/openwifi-1.3.1-wilsele-1-32bit.img.xz works correctly.

njqychenlin, 2023-03-24:
https://users.ugent.be/~xjiao/openwifi-1.3.1-wilsele-1-32bit.img.xz  why? I can not download the img file。

JiaoXianjun, 2023-03-24:
Please use our new img in readme

JiaoXianjun, 2023-03-27:
You should show the exact operations you did after downloading the img, and tell us all information/outputs (OS, version, SD card tool, board type, etc.).