open-sdr/openwifi #203: fail to run the monitor_ch.sh

xiaoxiaochuzm, 2022-08-16:
Hello, I encountered an error when running the monitor mode of CSI. The board  failed to get the CSI information correctly. However, in other modes, the board can work successfully.The printed information is as follows.
[image]
I've tried many times, but I still haven't succeeded,so I take the liberty to disturb you.
Look forward to your reply!

JiaoXianjun, 2022-09-22:
Can you check the trigger condition of the side channel?
Can you see any packet via tcpdump?

closed 2023-04-28 by JiaoXianjun

chenmo234, 2024-05-22:
Hello, sorry to bother you. May I ask you some questions regarding IQ capture in OpenWiFi? Here is my contact information: c2385405729.