open-sdr/openwifi #205: SDR with 802.11ax Feature

meena-devi, 2022-08-19:
Hi,

Thank you in advance.

1. As you have mentioned the 802.11ax feature is under development. May I please know if any update on it.

2. I wanted to run the code base on usrp b210 with 802.11 ax feature. So does the openwifi supports for SDR - usrp b210?

Regards,
meena-devi

JiaoXianjun, 2022-08-19:
1. We are working on it. It is quite complicated, so the progress is slower than our expectation
2. We never try b210. You need to figure out how to run/port it yourself.

meena-devi, 2022-08-22:
> 1. We are working on it. It is quite complicated, so the progress is slower than our expectation
> 
>     2. We never try b210. You need to figure out how to run/port it yourself.

Thank you so much for your response

closed 2022-09-14 by JiaoXianjun