open-sdr/openwifi #207: Perhaps iptables or your kernel needs to be upgraded

Weimiao0720, 2022-08-23:
0. Could you send email to xianjun.jiao@ugent.be to introduce your self?
not yet
1. Our image is used directly or you build your own image?
used directly
2. What is your own modification?
nothing
3. Versions: OS, Vivado, openwifi/openwifi-hw repo branch and commit revision
vivado
4. Board/hardware type
 ZCU102
5. WiFi channel number
1
6. Steps to reproduce the issue, and the related error message, screenshot, etc
[image]
7. Describe your debug efforts by Linux native tools, such as tcpdump and "cat /proc/interrupts"
root@analog:~# sudo iptables -t nat -A POSTROUTING -o eth0 -j MASQUERADE    
iptables v1.4.21: can't initialize iptables table `nat': No chain/target/match by that name
Perhaps iptables or your kernel needs to be upgraded.
8. Describe your debug efforts by: https://github.com/open-sdr/openwifi/blob/master/doc/README.md#Debug-methods

9. Any other thing we need to know for helping you better?

weiliu1011, 2022-08-23:
Hi Weimiao,
Did you execute this iptable command on the openwifi board or on your computer?

Weimiao0720, 2022-08-23:
> Hi Weimiao, Did you execute this iptable command on the openwifi board or on your computer?

I did it on board（ZCU102）

weiliu1011, 2022-08-23:
You should execute it on ur PC, the goal of this command is to enable NAT on your PC so it can give ur openwifi board Internet access. Please follow the quick start session carefully.

Weimiao0720, 2022-08-23:
> You should execute it on ur PC, the goal of this command is to enable NAT on your PC so it can give ur openwifi board Internet access. Please follow the quick start session carefully.

Thank you so much .
There is another error.
When I finished reading your article, I tried to reproduce wireless time synchronization.I recently tried to synchronize two blocks of Xilinx ZCU102 using Linux PTP, but ran into the following issues：
sudo ptp4l -i eth0 -m -S
ptp4l[4863.619]: port 1 (eth0): INITIALIZING to LISTENING on INIT_COMPLETE
ptp4l[4863.620]: port 0 (/var/run/ptp4l): INITIALIZING to LISTENING on INIT_COMPLETE
ptp4l[4863.620]: port 0 (/var/run/ptp4lro): INITIALIZING to LISTENING on INIT_COMPLETE
ptp4l[4863.620]: Switched to /dev/ptp0 as PTP clock
ptp4l[4863.620]: port 1 (eth0): LISTENING to FAULTY on FAULT_DETECTED (FT_UNSPECIFIED)
ptp4l[4879.621]: port 1 (eth0): FAULTY to LISTENING on INIT_COMPLETE
ptp4l[4885.646]: port 1 (eth0): LISTENING to MASTER on ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
ptp4l[4885.646]: selected local clock 000a35.fffe.049db5 as best master
ptp4l[4885.646]: port 1 (eth0): assuming the grand master role
ptp4l[4885.647]: port 1 (eth0): unexpected socket error
ptp4l[4885.648]: port 1 (eth0): MASTER to FAULTY on FAULT_DETECTED (FT_UNSPECIFIED)

LinuxPTP is installed under an open source image called openwifi on github. I suspect that the openwifi kernel or the ZCU102 port driver cause this error.

I really can't solve the problem. I will appreciate if you can help.THANK YOU SO SO MUCH.

weiliu1011, 2022-08-23:
We are not linuxptp expert, I have successully run linuxptp v2.0 on openwifi image but not the most recent one. I suggest we do not further discuss this subject here, as it is no longer related to the original issue.

closed 2022-09-22 by JiaoXianjun