open-sdr/openwifi #213: Some questions about Clock frequency and IFFT module.

adventurecs, 2022-09-07:
Hi~
Thanks Mr.Jiao in advanced.
I am using ZCU102 and fmcomms4.
I want to make sure about some entry-level quentions. I've read some code and documentation, but I'm still not sure about them.

1. What is the Clock Frequency in openofdm/dot11_tx.v ?   40M?

2. How many clock cycles does the IFFT core takes to complete the IFFT of an OFDM symbol？（maybe a round answer?）  Does it input the frequency domain data into IFFT module after finishing the output of the previous symbol?

3. Does it store the time domain data of all symbols in a buffer before send a packet? I think the speed of IFFT maybe not fast enough.

JiaoXianjun, 2022-09-14:
1. 100M. It is generated by clk_wiz_0 in the top level system.bd: Open the whole top level project via Vivado. Check how to do this in the openwifi-hw README.
2. I forgot. You can do a simulation of openofdm_rx/openofdm_tx to check the waveform in vivado. Check how to do this in the openwifi-hw README.
3. NO. the IFFT is fast enough. We do not store all time domain data at tx side.

adventurecs, 2022-09-21:
Thanks a lot!

closed 2022-09-24 by JiaoXianjun