open-sdr/openwifi #220: adi git clone stuck during ZC706 sdcard update

briansune, 2022-09-19:
Had Emailed Xianjun Jiao and discussed which open issue accordingly.

When running the sdcard update script, the adi-linux git pull stuck and unable to finish the script.
Which back to a simple question, are manually git cloning ADI Linux 2019_R1 makes no different?
Doing git clone website is a better method on the script rather than a linkage?

Thank you

JiaoXianjun, 2022-09-19:
Of course you can do it manually. Make sure you follow/understand: https://github.com/open-sdr/openwifi/blob/master/user_space/prepare_kernel.sh

briansune, 2022-09-19:
`cd $OPENWIFI_DIR/
git submodule init $LINUX_KERNEL_SRC_DIR_NAME
git submodule update $LINUX_KERNEL_SRC_DIR_NAME
cd $OPENWIFI_DIR/$LINUX_KERNEL_SRC_DIR_NAME
git checkout 2019_R1
git pull origin 2019_R1
git reset --hard
# git reset --hard 4e81f0927cfb2fada92fc762dbd65d002848405a
cp $LINUX_KERNEL_CONFIG_FILE ./.config
cp $OPENWIFI_DIR/driver/ad9361/ad9361.c $OPENWIFI_DIR/$LINUX_KERNEL_SRC_DIR_NAME/drivers/iio/adc/ad9361.c -rf
cp $OPENWIFI_DIR/driver/ad9361/ad9361_conv.c $OPENWIFI_DIR/$LINUX_KERNEL_SRC_DIR_NAME/drivers/iio/adc/ad9361_conv.c -rf`

As we can see the line after CD git it trying to clone the ADI repository, which the linkage might required update.
As this is not working while using the script.

JiaoXianjun, 2022-09-20:
Could you please elaborate which link is wrong and what is the correct link?

briansune, 2022-09-20:
>>>> git submodule init $LINUX_KERNEL_SRC_DIR_NAME
>>>> git submodule update $LINUX_KERNEL_SRC_DIR_NAME

Basically if the ADI Linux is clone whatever branch or tag it is grepped it can change via
git checkout 2019_R1
git pull origin 2019_R1

Long story short the base repository not even clone to local aka stuck and not able to run git checkout accordingly.

JiaoXianjun, 2022-09-20:
Can you manually git clone that adi repository?

briansune, 2022-09-20:
Yes, If manually override the script and clone before run the full SD Update script then it is solved.
So you could improve the git clone method in this part as linkage of git to another git could be problem buggy.
Or simplest way is just using ADI git website rather than super-link between repository?

JiaoXianjun, 2022-09-20:
Actually git submodule does the same thing as git clone. Maybe you encountered the network connection issue?

briansune, 2022-09-20:
I understand it is doing the same as git clone, but can this work on your side after few years of development. Make sure a fresh clean pure repository clone from open-wifi then run the same sd card update script. If no stuck then it could be a region issue about github.

JiaoXianjun, 2022-09-22:
I just tested. I don't have any issue.
The clone will take quite a while, so you need to be patient.

closed 2022-09-22 by JiaoXianjun

briansune, 2022-09-22:
Hello Jiao,

May I ask what amount of time you are talking about? I had waited 3hours at least before I report such issues.
While my network at least got 900Mbs up down speed.
So please provide more clue to my side to see if this is a region issue.

JiaoXianjun, 2022-09-22:
I didn't count. But definitely less than 3 hours. My side it is like 0.5 or < 1 hour.

briansune, 2022-09-22:
This is puzzled, as manually git clone 2019_R1 takes less than 5minutes.
So it is paramount that you update the submodule rather than the old repository of ADI.
Unless I missed anything the repository revision you are trying to clone with submodule is not important as after those line in the script it will grep 2019_R1.
So could updating the submodule to 2019_R1 reduce the chance of cloning issue?

JiaoXianjun, 2022-09-22:
What I did:

1. A clean clone of openwifi
2. prepare_kernel.sh XILINX_DIR 32

Above script will stay quite a while at clone operation, then all the rest executed automatically after clone finished.

briansune, 2022-09-22:
So this is not following SD update script but stand-alone prepare_kernel.sh
I will try any different, but the discrepancy between submodule and manually clone is huge.
B2U after testing.

JiaoXianjun, 2022-09-22:
They did the same thing:

https://github.com/open-sdr/openwifi/blob/a4e76cf69675486b837bd79c578a10f278b10409/user_space/update_sdcard.sh#L89

briansune, 2022-09-22:
Yes, understand as you want to reduce the sd card setup work load on your side.
Meantime, I think this is due to VPN issue.
I offed VPN and after 30 minutes it is compiling the kernel.
But compared to the manual clone it is still far far slower no idea why.

Problem solved, thank you