open-sdr/openwifi #223: Zedboard WiFi client issue

andreascas, 2022-09-22:
Hello,

I am running with the image provided for zedboard_fmcs2, with no modifications and the newest image available here.
My issue is that trying to run the openwifi implementation on the zedboard i cannot use the board as a client.

I have validated that it operates fine as an AP, by running the AP code `./wgd.sh` and `./fosdem.sh`. From my laptop i could then join the AP, openwifi, and browse to 192.168.13.1 and see the content, hence the AP functionality works.

For a client on the other hand i run the commands `./wgd.sh`, `route del default gw 192.168.10.1`, `wpa_supplicant -i sdr0 -c wpa-connect.conf &`, and here i find myself stuck trying to connect to my AP. I have tried with a password and with no password on it, and making the respective changes in wpa-connect.conf. This is the output i find in the terminal, which tells me it is an auth failure. 
[logs.txt](https://github.com/open-sdr/openwifi/files/9623963/logs.txt)

I am stuck, and hoping that you could provide guidance on the issue, and how i may proceed, as i would like to connect the zedboard as a client.

Thanks in advance
Andreas

JiaoXianjun, 2022-09-22:
2.4GHz is always tricky, due to the 11b compliant requirement. We do not have 11b. You might try following methods:
1. Limit your AP to OFDM only. No 11b. 
2. If not possible, restrict your AP to a 5GHz channel.
3. If you stick to 2.4GHz, you can try with (after your run wpa_supplicant, in another shell): ./set_lbt_th.sh 60 ( https://github.com/open-sdr/openwifi/blob/master/doc/app_notes/frequent_trick.md#cca-lbt-threshold-and-disable )
4. Also take a look at: https://github.com/open-sdr/openwifi#special-note-for-11b
5. Test without password

We will have a better version soon (maybe several weeks.)

NTUGY, 2022-10-20:
> Hello,
>
> I am running with the image provided for zedboard_fmcs2, with no modifications and the newest image available here. My issue is that trying to run the openwifi implementation on the zedboard [...]

I had the same problem.I set AP to 5GHz channel. As a client, Zedboard encounters the same situation as you

JiaoXianjun, 2023-02-01:
Now 2.4GHz works better than before: https://github.com/open-sdr/openwifi/releases/tag/v1.4.0

Engineer2223, 2023-06-05:
I’m unable to access internet.
I’m using openwifi on board ADRV9461-z7035. I connect my iphone openwifi after enabling AP (./wgd.sh and ./fosdem.sh). Browse to 192.168.13.1 is ok and video plays fine BUT i am unable to browse to any other website/video.

Wifi client internet access on PC output is as follows

cateyes@DESKTOP-RQO8CMH:~$ sudo sysctl -w net.ipv4.ip_forward=1
net.ipv4.ip_forward = 1
cateyes@DESKTOP-RQO8CMH:~$ sudo iptables -t nat -A POSTROUTING -o WIFI -j MASQUERADE
cateyes@DESKTOP-RQO8CMH:~$ sudo ip route add 192.168.13.0/24 via 192.168.10.122 dev eth0
Error: Nexthop has invalid gateway.
cateyes@DESKTOP-RQO8CMH:~$

Please guide.