open-sdr/openwifi #224: Open Wifi AD9361 CMOS interface

briansune, 2022-09-28:
Background ZC706-based lower dev board cost experiment.

I would like to make sure few things about the interface and the requirements.

Q1, AD9361 CMOS mode single channel allow the hardware to use maximum 61.44M BW.
While Open-WIFI require 40M BW so can open-WIFI only use 1 set of TX RX?

Q2 This is about some warning after rebuild.
During running the ./wgd.sh
depmod: WARNING: could not open /lib/modules/4.14.0-gb6e379910a11-dirty/modules.order: No such file or directory
depmod: WARNING: could not open /lib/modules/4.14.0-gb6e379910a11-dirty/modules.builtin: No such file or directory

Thank you

JiaoXianjun, 2022-09-28:
Q1: In principle yes. But I do not remember. Maybe you can find some info here: https://github.com/open-sdr/openwifi/issues/99 and discuss with @regymm

Q2: It does not matter

regymm, 2022-09-28:
Hi, I haven't used CMOS mode(in the Vivado block design) so far. But One thing is you can use external level shifter chips to convert between LVDS and single-ended signals. Then in verilog code you can add some simple coupling logic, while the main module remains unaware of the physical signals. But of course, if you convert too much and have poor PCB routing the BW will be compromised.

JiaoXianjun, 2022-09-28:
If I am correct, we always use 2tx2rx mode. I forgot whether I tried 1tx1rx for @regymm before. (I tried?)

regymm, 2022-09-29:
In the Vivado block design, the ad9361 module parameter has to remain in 2tx2rx mode(otherwise package timing will be wrong), but in the Linux device tree using 1tx1rx is also fine.

briansune, 2022-09-29:
@JiaoXianjun 

A short update about the 1TXRX, not sure why RX keep 0:
[image]

I just modify the dtb in the SD card to enable CMOS and 1 RXTX mode.
Antenna is attached 2.4G specification.

briansune, 2022-09-30:
> If I am correct, we always use 2tx2rx mode. I forgot whether I tried 1tx1rx for @regymm before. (I tried?)

@JiaoXianjun 
https://github.com/open-sdr/openwifi/issues/123#issuecomment-984446047

I think you do try 1TXRX mode working. So I will cont'd debug a bit on my side.

Thank you

briansune, 2022-10-05:
@JiaoXianjun 

May how can I resolve the following debug message?

00:04:28.511735 67270190us tsft bad-fcs 18.0 Mb/s 2437 MHz 11g -66dB signal antenna 1 unknown IEEE802.11 frame type (3)(header) unknown IEEE802.11 frame type (3)unknown 802.11 frame type (3)
00:04:28.512666 67271049us tsft bad-fcs 24.0 Mb/s 2437 MHz 11g -66dB signal antenna 1 ATIM
00:04:28.512816 67271289us tsft bad-fcs 54.0 Mb/s 2437 MHz 11g -66dB signal antenna 1 unknown IEEE802.11 frame type (3)(header) unknown IEEE802.11 frame type (3)unknown 802.11 frame type (3)
00:04:28.513220 67271508us tsft bad-fcs 36.0 Mb/s 2437 MHz 11g -66dB signal antenna 1 Power Save-Poll AID(859b)
00:04:28.521530 67279803us tsft bad-fcs 48.0 Mb/s 2437 MHz 11g -66dB signal antenna 1 unknown IEEE802.11 frame type (3)(header) unknown IEEE802.11 frame type (3)unknown 802.11 frame type (3)
00:04:28.529293 67287680us tsft bad-fcs 12.0 Mb/s 2437 MHz 11g -65dB signal antenna 1 Request-To-Send TA:97:4d:82:c5:be:c1 (oui Unknown)
00:04:28.535045 67293384us tsft bad-fcs 54.0 Mb/s 2437 MHz 11g -66dB signal antenna 1 CF +QoS
00:04:28.543924 67302023us tsft bad-fcs 24.0 Mb/s 2437 MHz 11g -66dB signal antenna 1 unknown IEEE802.11 frame type (3)(header) unknown IEEE802.11 frame type (3)unknown 802.11 frame type (3)
00:04:28.557557 67315972us tsft bad-fcs 12.0 Mb/s 2437 MHz 11g -66dB signal antenna 1 Authentication (Reserved)-365a: [|802.11]
00:04:28.558970 67316615us tsft bad-fcs 9.0 Mb/s 2437 MHz 11g -66dB signal antenna 1 CF Ack/Poll+QoS
00:04:28.573688 67332146us tsft bad-fcs 48.0 Mb/s 2437 MHz 11g -65dB signal antenna 1 Unknown Ctrl Subtype
00:04:28.574956 67332635us tsft bad-fcs 12.0 Mb/s 2437 MHz 11g -66dB signal antenna 1 Control Wrapper
00:04:28.575151 67333517us tsft bad-fcs 36.0 Mb/s 2437 MHz 11g -66dB signal antenna 1 Unknown Ctrl Subtype
00:04:28.580279 67338742us tsft bad-fcs 36.0 Mb/s 2437 MHz 11g -66dB signal antenna 1 CF Poll+QoS

closed 2022-10-22 by briansune