open-sdr/openwifi #245: Compile the Openwifi kernel

xueduanyue, 2022-11-06:
Hello,
I want to compile the Linux kernel under zed_fmcs2.
As a preliminary test, I used an unmodified kernel to test if the compilation was successful.
After this command:
./prepare_kernel.sh $XILINX_DIR 32 build
I can find the Linux kernel in : adi-linux/arch/arm/boot/uImage
I manually replaced the uImage file to the boot directory
And I install the driver on thet board using this command:
cd openwifi/driver; scp `find ./ -name \*.ko` root@192.168.10.122:openwifi/

I get an error when I start wgd.sh for testing:
The sdr.ko seems to be failed.
Can someone tell me how to solve this problem？
Or a tutorial on replacing the kernel？
Thanks for all help.

Here is the output:
rmmod: ERROR: Module tx_intf is not currently loaded
+ insmod .//tx_intf.ko
tx_intf is loaded!
rmmod: ERROR: Module rx_intf is not currently loaded
+ insmod .//rx_intf.ko
rx_intf is loaded!
rmmod: ERROR: Module openofdm_tx is not currently loaded
+ insmod .//openofdm_tx.ko
openofdm_tx is loaded!
rmmod: ERROR: Module openofdm_rx is not currently loaded
+ insmod .//openofdm_rx.ko
openofdm_rx is loaded!
rmmod: ERROR: Module xpu is not currently loaded
+ insmod .//xpu.ko
xpu is loaded!
rmmod: ERROR: Module sdr is not currently loaded
+ insmod .//sdr.ko test_mode=0
insmod: ERROR: could not insert module .//sdr.ko: Unknown symbol in module
sdr is not loaded!

JiaoXianjun, 2022-11-08:
You need to describe what have you done from the very beginning. From when you take a blank SD card. Only after seeing that we can guess what is wrong.

xueduanyue, 2022-11-08:
> 
OK.I will sort out the steps
1.Get a blank SD card and a zed_fmcs2 board.
2.Download the openwifi-1.3.1-wilsele-1-32bit.img.
3.Follow the Quick start, and replace the BOOT,devicetree and uImage.
4.Now I test the Openwifi. Everything is OK.I can see and connect openwifi and open the test web.

5.Prepare a Ubuntu for Compile. And Install dependent libraries.
6.Download the openwifi-master and adi-linux. Move adi-linux into openwifi-master.
7.Switch the terminal to the user_space directory. Compile the Linux kernel.
./prepare_kernel.sh $XILINX_DIR 32 build
8.After compiling, in adi-linux/arch/arm/boot, I find the image,uImage and zImage. 
9.Copy the uImage to SD card.
9.Switch the terminal to the driver directory. Compile the driver.
./make_all.sh $XILINX_DIR 32
10.Copy the driver files to the board via ssh channel
cd openwifi/driver; scp `find ./ -name \*.ko` root@192.168.10.122:openwifi/
11.Now I test the kernel.
The new kernel can be loaded successfully. I can log in via ssh as usual.
When I run wgd.sh, I get the above error.

A-T-Kristensen, 2022-11-08:
Hi

When you run wgd.sh, what does dmesg return? That should tell you which symbol is unknown in sdr.ko

xueduanyue, 2022-11-08:
@A-T-Kristensen @JiaoXianjun 
Thanks for your help. I check the dmesg.

I meet a problem with Unknown symbol.
This [Build openwifi Linux img from scratch] solution seems to work for me.
But the 2019_R1-2020_06_22.img.xz file appears to be invalid.
Is there any other suitable version or a download link?

The dmesg prints as follows:
sdr,xpu dev_probe reset tsf timer
sdr,xpu dev_probe XPU_REG_TSF_RUNTIME_VAL_LOW_read 1 33497 100513us
sdr,xpu dev_probe succeed!
sdr,xpu hw_init mode 1
sdr,xpu hw_init mode XPU_NORMAL
sdr,xpu hw_init err 0
sdr: Unknown symbol ieee80211_queue_stopped (err 0)
sdr: Unknown symbol ieee80211_start_tx_ba_cb_irqsafe (err 0)
sdr: no symbol version for openofdm_rx_api
sdr: Unknown symbol ieee80211_free_hw (err 0)
sdr: Unknown symbol ieee80211_register_hw (err 0)
sdr: Unknown symbol ieee80211_ctstoself_duration (err 0)
sdr: Unknown symbol ieee80211_wake_queue (err 0)
sdr: Unknown symbol ieee80211_tx_status_irqsafe (err 0)
sdr: Unknown symbol wiphy_rfkill_set_hw_state (err 0)
sdr: Unknown symbol __cfg80211_alloc_reply_skb (err 0)
sdr: Unknown symbol ieee80211_stop_tx_ba_cb_irqsafe (err 0)
sdr: Unknown symbol wiphy_rfkill_stop_polling (err 0)
sdr: Unknown symbol wiphy_apply_custom_regulatory (err 0)
sdr: Unknown symbol ieee80211_alloc_hw_nm (err 0)
sdr: Unknown symbol cfg80211_vendor_cmd_reply (err 0)
sdr: Unknown symbol ieee80211_stop_queue (err 0)
sdr: Unknown symbol wiphy_rfkill_start_polling (err 0)
sdr: Unknown symbol ieee80211_unregister_hw (err 0)
sdr: Unknown symbol ieee80211_beacon_get_tim (err 0)
sdr: Unknown symbol ieee80211_rx_irqsafe (err 0)

A-T-Kristensen, 2022-11-08:
Hi

Please see this part https://github.com/open-sdr/openwifi#build-openwifi-linux-img-from-scratch

I believe you may be missing some steps from there like the step running [post_config.sh](https://github.com/open-sdr/openwifi/blob/master/user_space/post_config.sh) which does `modprobe mac80211` which should give you the missing symbols.

xueduanyue, 2022-11-09:
@A-T-Kristensen 
Yes, I did miss some steps.I will check this part carefully again.

Here is another question in this part:
There is a step in this part https://github.com/open-sdr/openwifi#build-openwifi-linux-img-from-scratch:
Download 2019_R1-2020_06_22.img.xz from Analog Devices Wiki. Burn it to a SD card.
But the 2019_R1-2020_06_22.img.xz file appears to be invalid.
Is there any other suitable version or a download link?

A-T-Kristensen, 2022-11-09:
Hi

For downloading 2019_R1-2020_06_22.img.xz, the link on the analog webpage seems to not do anything for some reason, but if you just copy paste the link itself and put it in your browser it works (don't click on the link, copy-paste it in), the link is http://swdownloads.analog.com/cse/2019_R1-2020_06_22.img.xz 

If you click on the link in this github message it also does not do anything. **Copy paste like shown below works**

[image]

xueduanyue, 2022-11-09:
@A-T-Kristensen 
Thanks a lot for your help.
I can download it.
And this issue can be closed.

closed 2022-11-09 by xueduanyue

chenmo234, 2024-06-04:
I also need to modify sdr.c and perform the same operations as you. Can I discuss this with you? Here is my contact information: c2385405729.

> 非常感谢您的帮助。我可以下载它。这个问题可以解决。