open-sdr/openwifi #251: The packet loss rate of AP and STA is up to 3.7%，Is this a normal phenomenon？

NTUGY, 2022-11-13:
0. Could you send email to xianjun.jiao@ugent.be to introduce your self?
I am a student from Chongqing University of Posts and Telecommunications.
1. Our image is used directly or you build your own image?
I use your image directly.
2. What is your own modification?
None.
3. Versions: OS, Vivado, openwifi/openwifi-hw repo branch and commit revision
ubuntu18/Vivado2018.3/master branch 
4. Board/hardware type
Antsdr 
5. WiFi channel number
44
6. Steps to reproduce the issue, and the related error message, screenshot, etc
Two antsdr development boards are used as AP and STA respectively. I use iperf to test packet loss rate and bandwidth between AP and STA. The TX and RX antennas of the two development boards are orthogonal. And the distance between two development boards is 3cm. Packet loss rate is up to 3.7% !!!  Is this a normal phenomenon？
[image]

The command I use is as follows：
STA：iperf -s -p 6100 -u -i 1
AP  :  iperf -c  192.168.13.5 -p 6100 -u -i 1 -t 20 -b 40m

NTUGY, 2022-11-20:
> 
Hi Dr. Jiao，
However, if the Antsdr development board is used as the AP and the mobile phone is used as the STA, the packet loss rate measured with the iperf is very low, and the packet loss rate is only about 0.3%.

NTUGY, 2022-11-20:
> Seems OK. It is also environment dependent. You can try to tune the system according to https://github.com/open-sdr/openwifi/blob/master/doc/app_notes/frequent_trick.md

Hi Dr. Jiao，
However, if the Antsdr development board is used as the AP and the mobile phone is used as the STA, the packet loss rate measured with the iperf is very low, and the packet loss rate is only about 0.3%.

NTUGY, 2022-11-20:
> Seems OK. It is also environment dependent. You can try to tune the system according to https://github.com/open-sdr/openwifi/blob/master/doc/app_notes/frequent_trick.md

I checked RX data statistics and found that many packets were lost due to FCS failure

NTUGY, 2022-11-20:
Hi Dr. Jiao，
Antsdr is used as AP and STA respectively, and cables are used to connect them. I checked the RX data statistics. I found that many packets were lost due to the failure of FCS, so that data retransmission failed five times. This is the reason for the high packet loss rate. I want to know the reason for the failure of FCS. Is AD9361 not good enough as the front end?
[image]
[image]

JiaoXianjun, 2022-11-20:
Please play with all these parameters: https://github.com/open-sdr/openwifi/blob/master/doc/app_notes/frequent_trick.md

NTUGY, 2022-11-23:
> Please play with all these parameters: https://github.com/open-sdr/openwifi/blob/master/doc/app_notes/frequent_trick.md
Hi Dr. Jiao，
I carefully read the content of the link you gave me. When I override the Linux automatic control for ht packet, the packet loss rate is close to 0%. But  when I use the Linux rate adaptation algorithm minstrel_ht to set the packet rate/MCS automatically via openwifi_tx() function, the packet loss rate is close to 4%. 

I have another question. 
 ./sdrctl dev sdr0 set reg drv_tx 1 N
Does the command above  only change the data frame rate?Does it affect the management frame and control frame rate?
Do you have any good suggestions？

JiaoXianjun, 2022-12-08:
https://github.com/open-sdr/openwifi/blob/b147dfd1528021e018580b2fbcf1c7ed43239962/driver/sdr.c#L1087

closed 2023-01-04 by NTUGY