open-sdr/openwifi #252: Couldn't get openwifi to work at 2.4Ghz

KrisYuFi, 2022-11-22:
0. Could you send email to xianjun.jiao@ugent.be to introduce your self?
I've already sent it.
1. Our image is used directly or you build your own image?
I used your image.
2. What is your own modification?
Hello, Dr. Jiao, I have been trying to make the openwifi board work in different frequencies recently, and I want it to work in the AP mode of 2.4Ghz. So I modified the _openwifi/hostpad-openwifi.conf_ on the board:
before:
`interface=sdr0
driver=nl80211
country_code=BE
ssid=openwifi
hw_mode=a
channel=44
supported_rates=60 90 120 180 240 360 480 540
basic_rates=60 90 120 180
ieee80211n=1
#ht_capab=[SHORT-GI-20]
require_ht=1`
after:
`interface=sdr0
driver=nl80211
country_code=BE
ssid=openwifi
hw_mode=g
channel=9
supported_rates=60 90 120 180 240 360 480 540
basic_rates=60 90 120 180
ieee80211n=1
#ht_capab=[SHORT-GI-20]
require_ht=1`
I just modifed the hw_mode and channel. After I run the `./wgd.sh` and `./fosdem.sh`  ,I  couldn't see the 'openwifi' on my phone , sometimes the I could see it ,but it just disappeared when I try to connect to it. 
I also tried using ./fosdem-11ag.sh and modifing the 'hostpad-openwifi-11ag.conf' , but it also failed.
I compared./fosdem-11ag.sh with./fosdem.sh and hostpad-openwifi-11ag.conf with hostpad-openwifi.conf. The code is the same in the version 1.3.1,.
So, does this 
`(Use "./wgd.sh 1" to enable experimental AMPDU aggregation on top of 11n)
(Use "./fosdem-11ag.sh" to force 11a/g mode)` in the README/QUIK START  works?

4. Board/hardware type
zynq7020 ad9361

KrisYuFi, 2022-12-05:
WIFI in the 2.4G band can work properly, but it needs to be in a good electromagnetic environment，

closed 2022-12-05 by KrisYuFi

JiaoXianjun, 2023-02-01:
Now it works better than before: https://github.com/open-sdr/openwifi/releases/tag/v1.4.0

winfelin, 2023-11-25:
Why openwifi needs to be in a good electromagnetic environment when working in the 2.4G band?