open-sdr/openwifi #254: FPGA interal loopback

Fei8686, 2022-12-07:
Hi,
Thank you very much for providing this project, it helps a lot during my study. I am experimenting with a zc706_fmcs2 board.The image file I use is “openwifi-1.2.1-leuven-2-32bit.img”.The version of VIVADO I use is 2018.3.

In the app_notes,I noticed that you have proposed a test process related to self loopback(app_notes/packet-iq-self-loopback-test.md), including over-the-air loopback and FPGA internal loopback.

First of all,I connected the TX and RX of the board through the RF cable with 25dB attenuation.Then,I tried to test over-the-air loopback with the method you proposed,and it works.However,when I replace "./side_ch_ctl wh5h0" with “./side_ch_ctl wh5h4” to test FPGA internal loopback , the system does not work.The error is that when I injected the packets,the count of the outputs did not grow as expected.

My debug efforts are as follows:

· Open a terminal on the computer, and then in the terminal:

```
ssh root@192.168.10.122
cd openwifi
./wgd.sh
./monitor_ch.sh sdr0 44
./sdrctl dev sdr0 set reg xpu 8 1000
insmod side_ch.ko iq_len_init=8187
./side_ch_ctl wh11d100
./side_ch_ctl wh8d16
./sdrctl dev sdr0 set reg xpu 1 1
./side_ch_ctl wh5h4
./side_ch_ctl g0

```
· Open a new ssh session to the board from your computer. Then run on board:
```
cd openwifi/inject_80211/
make
./inject_80211 -m n -r 5 -n 1 sdr0

```
In the previous ssh session, the count becomes 1 as normal.Unfortunately,the count will grow by 1 only the first time I sent a packet. After that,no matter how many packets I sent, the count did not grow as expected.
If i once stop the "./side_ch_ctl g0" in the very first ssh session,rerun "./side_ch_ctl g0" again and do the packet injection in another ssh.The count is always 0.

· To avoid the system runs into dead state, I tried to put the receiver into reset state
`./sdrctl dev sdr0 set reg rx 0 1`
Then,do single packet Tx again
`./inject_80211/inject_80211 -m n -r 5 -n 1 sdr0`
But the count in the 1st ssh session is still 0.

Could you help me to solve the problem of how to carry out FPGA internal loopback in the board?

Thanks,
The shy

JiaoXianjun, 2022-12-07:
This is an known issue. Maybe it will be solved in the next release around the year end.

Fei8686, 2023-01-06:
Hi,
I'm sorry to bother you.Could you kindly let me know whether this issue is solved?

JiaoXianjun, 2023-01-28:
It works now:

https://github.com/open-sdr/openwifi/releases/tag/v1.4.0

closed 2023-03-30 by JiaoXianjun