open-sdr/openwifi #258: ADRV9361-Z7035 throughput

zhangsd20, 2022-12-29:
Performance (best case: aggregation/AMPDU on):

    iperf: TCP 40~50Mbps; UDP 50Mbps
    EVM -38dB; MCS0 sensitivity -87dBm; MCS7 -72dBm. (FMCOMMS2 2.4GHz; cable and OTA test)

The best result described is iperf tcp at 40~50Mbps, my actual test whether with the instrument test or I use the terminal to connect ap can not reach this bandwidth.

my test:

[  3] local 192.168.13.1 port 56380 connected with 192.168.13.7 port 5001
[ ID] Interval       Transfer     Bandwidth
[  3]  0.0- 1.0 sec  3.00 MBytes  25.2 Mbits/sec
[  3]  1.0- 2.0 sec  3.00 MBytes  25.2 Mbits/sec
[  3]  2.0- 3.0 sec  3.12 MBytes  26.2 Mbits/sec
[  3]  3.0- 4.0 sec  3.25 MBytes  27.3 Mbits/sec
[  3]  4.0- 5.0 sec  3.12 MBytes  26.2 Mbits/sec
[  3]  5.0- 6.0 sec  3.25 MBytes  27.3 Mbits/sec

I used an Analog device ADRV9361-Z7035 for the image.
The branch of openwifi is master.

What might be the cause of this?
what should i do ?

JiaoXianjun, 2023-01-04:
You can consider to enable AMPDU as we mentioned in the quick start guide:

Use "./wgd.sh 1" to enable experimental AMPDU aggregation on top of 11n

zhangsd20, 2023-01-05:
I've tested enable AMPDU ,enabling AMPDU throughput looks lower.
./wgd.sh 1
./fosdem.sh
test:
[  3] local 192.168.13.1 port 44054 connected with 192.168.13.7 port 5001
[ ID] Interval       Transfer     Bandwidth
[  3]  0.0- 1.0 sec   128 KBytes  1.05 Mbits/sec
[  3]  1.0- 2.0 sec   128 KBytes  1.05 Mbits/sec

JiaoXianjun, 2023-01-05:
For TCP, it is not stable (suspect compatibility issue with some clients). You could try UDP.
And you could also adjust the antenna, distance, etc.. You can also play with these tricks: https://github.com/open-sdr/openwifi/blob/master/doc/app_notes/frequent_trick.md

7035 has very low Tx power in 5GHz (ADI's design flaw), so I suggest you disconnect the Rx antenna (to avoid saturation caused by too close client), keep the Tx antenna, put the client (for example, the cell phone) very very close to the board (antenna/port).

zhangsd20, 2023-01-05:
UDP:
[  3] local 192.168.13.1 port 45442 connected with 192.168.13.4 port 5001
[ ID] Interval       Transfer     Bandwidth
[  3]  0.0- 1.0 sec  5.63 MBytes  47.3 Mbits/sec
[  3]  1.0- 2.0 sec  5.86 MBytes  49.2 Mbits/sec
[  3]  2.0- 3.0 sec  5.91 MBytes  49.6 Mbits/sec
[  3]  3.0- 4.0 sec  5.56 MBytes  46.7 Mbits/sec
[  3]  4.0- 5.0 sec  5.50 MBytes  46.1 Mbits/sec
[  3]  5.0- 6.0 sec  5.51 MBytes  46.2 Mbits/sec
[  3]  6.0- 7.0 sec  5.75 MBytes  48.2 Mbits/sec
[  3]  7.0- 8.0 sec  5.90 MBytes  49.5 Mbits/sec
[  3]  8.0- 9.0 sec  5.97 MBytes  50.0 Mbits/sec

But how can you configure it to improve tcp throughput?

zhangsd20, 2023-01-29:
Hi,

iperf: TCP 40~50Mbps; UDP 50Mbps.
The highest throughput I tested TCP was only 27Mbbps, and I never reached the 40Mbps described above.

If I want to improve tcp throughput,How to configure ?

JiaoXianjun, 2023-02-05:
I just tried our latest img with iperf3 between openwifi board and iPad (TCP). It is not that stable, but it does achieve 40~50Mbps with a big chance. Video: https://youtube.com/shorts/0TAqlmuysyo

So my suggestions:
1. Use our latest openwifi img.
2. Use Tx antenna only, and unplug the Rx antenna to give it a "CLEAN" environment from the receiver point of view.
3. Put the client VERY (as close as possible) close to the openwifi board, because openwifi Tx power is much lower than normal WiFi router.
4. IMPORTANT: Find the unused CLEAN WiFi channel at your location/environment (or the least busy channel). 5GHz channel is preferred.
5. Command on openwifi board: "wgd.sh 1" and "fosdem.sh"
6. Command on openwifi board: "iperf3 -s -i1"
7. App on iPad: iPerf 3 Wifi Speed Test (Server address 192.168.13.1, port 5201, Download, 5 Streams, Test duration 10s)

But due to the fact that different boards have different frequency offsets (on board crystal difference), it is normal that if you see the different performance. Our algorithm has limited performance while handling big frequency offset. We are still working on it.

Good luck!

commits:
openwifi https://github.com/open-sdr/openwifi/commit/a0972944154fce74e9db0908970c2bb2f8a31978
openwifi-hw https://github.com/open-sdr/openwifi-hw/commit/691e9d86b788747ed3de74cb2c8793f94b5fff62
openofdm https://github.com/open-sdr/openofdm/commit/747101dca8d00b4d0a99bd07dba647cd49948d04

zhangsd20, 2023-02-17:
I'm sorry for taking so long to get back to you.

Vivado 2021.1 was reinstalled as required and a new img was used.

No matter how I configure or adjust the device location, the TCP throughput can never exceed 30Mbps.

test:
root@analog:~/openwifi# iperf3 -c 192.168.13.4 -i 1 -t 60 -P 5
Connecting to host 192.168.13.4, port 5201
[  5] local 192.168.13.1 port 42206 connected to 192.168.13.4 port 5201
[  7] local 192.168.13.1 port 42208 connected to 192.168.13.4 port 5201
[  9] local 192.168.13.1 port 42210 connected to 192.168.13.4 port 5201
[ 11] local 192.168.13.1 port 42212 connected to 192.168.13.4 port 5201
[ 13] local 192.168.13.1 port 42214 connected to 192.168.13.4 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  1.05 MBytes  8.84 Mbits/sec    0   83.4 KBytes       
[  7]   0.00-1.00   sec   895 KBytes  7.33 Mbits/sec    0   72.1 KBytes       
[  9]   0.00-1.00   sec   485 KBytes  3.97 Mbits/sec    3   26.9 KBytes       
[ 11]   0.00-1.00   sec   648 KBytes  5.30 Mbits/sec    1   43.8 KBytes       
[ 13]   0.00-1.00   sec   844 KBytes  6.91 Mbits/sec    1   45.2 KBytes       
[SUM]   0.00-1.00   sec  3.86 MBytes  32.4 Mbits/sec    5             
- - - - - - - - - - - - - - - - - - - - - - - - -
[  5]   1.00-2.00   sec   764 KBytes  6.25 Mbits/sec    2   53.7 KBytes       
[  7]   1.00-2.00   sec   636 KBytes  5.21 Mbits/sec    2   43.8 KBytes       
[  9]   1.00-2.00   sec   399 KBytes  3.26 Mbits/sec    0   29.7 KBytes       
[ 11]   1.00-2.00   sec   452 KBytes  3.70 Mbits/sec    2   26.9 KBytes       
[ 13]   1.00-2.00   sec   636 KBytes  5.21 Mbits/sec    0   59.4 KBytes       
[SUM]   1.00-2.00   sec  2.82 MBytes  23.6 Mbits/sec    6

JiaoXianjun, 2023-04-28:
Normally this kind of SoM has worse performance than the independent RF card like fmcomms2/3/4.
I don't know why. Maybe due to the on board crystal.

closed 2023-04-28 by JiaoXianjun

jaimeMA5, 2025-03-04:
> UDP: [ 3] local 192.168.13.1 port 45442 connected with 192.168.13.4 port 5001 [ ID] Interval Transfer Bandwidth [ 3] 0.0- 1.0 sec 5.63 MBytes 47.3 Mbits/sec [ 3] 1.0- 2.0 sec 5.86 MBytes 49.2 Mbits/sec [ 3] 2.0- 3.0 sec 5.91 MBytes 49.6 Mbits/sec [ 3] 3.0- 4.0 sec 5.56 MBytes 46.7 Mbits/sec [ 3] 4.0- 5.0 sec 5.50 MBytes 46.1 Mbits/sec [ 3] 5.0- 6.0 sec 5.51 MBytes 46.2 Mbits/sec [ 3] 6.0- 7.0 sec 5.75 MBytes 48.2 Mbits/sec [ 3] 7.0- 8.0 sec 5.90 MBytes 49.5 Mbits/sec [ 3] 8.0- 9.0 sec 5.97 MBytes 50.0 Mbits/sec
> 
> If I want to improve tcp throughput,How to configure ?

Hi @zhangsd20, I'd like to know which parameters from the frequent_tricks page did you modify to reach the 50 Mbps in UDP's throughput (apart from AMPDU on). Thanks in advance!