open-sdr/openwifi #266: CSI fuzzer

mkk0719, 2023-01-16:
Hello, I'm learning the process of the csi fuzzer. Can you tell me in the linux kernal space, is sdr.c call the module tx_intf.c and send the value to the fpga? Can you tell me how the value of the csi fuzzer is written to the fpga register?

JiaoXianjun, 2023-01-24:
https://github.com/open-sdr/openwifi/blob/master/doc/app_notes/csi_fuzzer.md

mkk0719, 2023-01-25:
Thank you, I have read csi_fuzzer.md. And in the userspace, you use sdrctl and cmd.c to send the value of csi fuzzer by netlink to the linux kernal space, I want to know which code segment do the linux kernal space receive netlink. Is it in the function side_ch_nl_recv_msg of the side_ch.c?

JiaoXianjun, 2023-01-25:
Have you read our project document: https://github.com/open-sdr/openwifi/blob/master/doc/README.md

mkk0719, 2023-01-29:
Thank you very much, I have learned it from the function static openwifi_testmode_cmd in the sdrctl_intf.c, case REG_CMD_SET.

closed 2023-02-01 by JiaoXianjun