open-sdr/openwifi #273: Is there something wrong with your img file?

zhangmazi123321, 2023-02-10:
Hello, I downloaded your latest img file for booting zc702, but I can't enter commands in SecureCRT after booting. I changed to ADI's official img file and can enter commands normally. I can also use the previous version of img file normally. Is there something wrong with your img file?

JiaoXianjun, 2023-02-10:
Can you describe what have you done after you download the img file?

JiaoXianjun, 2023-02-10:
Can you also connect the UART/serial to monitor the output during booting?

zhangmazi123321, 2023-02-10:
Just follow your quick start for SD card boot, nothing superfluous. The startup information can be displayed normally, but the command cannot be entered.

JiaoXianjun, 2023-02-10:
Can you paste your UART output?

you can check your ethernet configuration via the console on UART

zhangmazi123321, 2023-02-10:
I connected through a serial port. The boot information is as follows

U-Boot 2014.07-dirty (Nov 20 2014 - 17:07:55)

Board:  Xilinx Zynq
I2C:   ready
DRAM:  ECC disabled 1 GiB
MMC:   zynq_sdhci: 0
SF: Detected N25Q128A with page size 256 Bytes, erase size 64 KiB, total 16 MiB
*** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
Net:   Gem.e000b000
Hit any key to stop autoboot:  0 
Device: zynq_sdhci
Manufacturer ID: 9f
OEM: 5449
Name: SD32G 
Tran Speed: 50000000
Rd Block Len: 512
SD version 3.0
High Capacity: Yes
Capacity: 28.8 GiB
Bus Width: 4-bit
reading uEnv.txt
407 bytes read in 30 ms (12.7 KiB/s)
Loaded environment from uEnv.txt
Importing environment from SD ...
Running uenvcmd ...
Copying Linux from SD to RAM...
reading uImage
5593704 bytes read in 479 ms (11.1 MiB/s)
reading devicetree.dtb
21003 bytes read in 38 ms (539.1 KiB/s)
reading uramdisk.image.gz
** Unable to read file uramdisk.image.gz **
## Booting kernel from Legacy Image at 03000000 ...
   Image Name:   Linux-5.10.0-98248-g1bbe32fa5182
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    5593640 Bytes = 5.3 MiB
   Load Address: 00008000
   Entry Point:  00008000
   Verifying Checksum ... OK
## Flattened Device Tree blob at 02a00000
   Booting using the fdt blob at 0x2a00000
   Loading Kernel Image ... OK
   Loading Device Tree to 1fff7000, end 1ffff20a ... OK

Starting kernel ...

C:0x000080C0-0x0055DA40->0x010C9800-0x0161F180
Uncompressing Linux... done, booting the kernel.
Booting Linux on physical CPU 0x0
Linux version 5.10.0-98248-g1bbe32fa5182-dirty (xjiao@wh-laptop) (arm-xilinx-linux-gnueabi-gcc.real (GCC) 10.2.0, GNU ld (GNU Binutils) 2.35.0.20200730) #3 SMP PREEMPT Sat Feb 4 13:48:45 CET 2023
CPU: ARMv7 Processor [413fc090] revision 0 (ARMv7), cr=18c5387d
CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
OF: fdt: Machine model: Xilinx Zynq ZC702
OF: fdt: earlycon: stdout-path /amba@0/uart@E0001000 not found
Memory policy: Data cache writealloc
cma: Reserved 128 MiB at 0x38000000
Zone ranges:
  Normal   [mem 0x0000000000000000-0x000000002fffffff]
  HighMem  [mem 0x0000000030000000-0x000000003fffffff]
Movable zone start for each node
Early memory node ranges
  node   0: [mem 0x0000000000000000-0x000000003fffffff]
Initmem setup node 0 [mem 0x0000000000000000-0x000000003fffffff]
percpu: Embedded 15 pages/cpu s29900 r8192 d23348 u61440
[... 404 lines cut ...]
[  OK  ] Started /etc/rc.local Compatibility.
[  OK  ] Started HTTP based time synchronization tool.
[  OK  ] Finished Permit User Sessions.
         Starting Light Display Manager...
         Starting Hold until boot process finishes up...
[  OK  ] Started CUPS Scheduler.
[  OK  ] Started Make remote CUPS printers available locally.
[  OK  ] Started LSB: Webfs simple HTTP server.
[FAILED] Failed to start VNC Server for X11.
[FAILED] Failed to start LSB: DHCP server.

Raspbian GNU/Linux 11 analog ttyPS0

analog login: root (automatic login)

Linux analog 5.10.0-98248-g1bbe32fa5182-dirty #3 SMP PREEMPT Sat Feb 4 13:48:45 CET 2023 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Thu Feb  9 10:45:28 GMT 2023 on ttyPS0
root@analog:~#

JiaoXianjun, 2023-02-10:
then run “ifconfig”
And
Ping your computer.

zhangmazi123321, 2023-02-10:
That's what I said earlier, it can't enter commands

JiaoXianjun, 2023-02-10:
ok you mean it freezes? Halt?

zhangmazi123321, 2023-02-10:
yes

JiaoXianjun, 2023-02-10:
[devicetree.dtb.zip](https://github.com/open-sdr/openwifi/files/10707184/devicetree.dtb.zip)

Can you unzip this devicetree and put devicetree.dtb into your SD card BOOT partition? It should solve your issue. Let me know it works or not. Thanks!

zhangmazi123321, 2023-02-10:
Woohoo, it worked!

JiaoXianjun, 2023-02-10:
I just commit the change: https://github.com/open-sdr/openwifi/commit/fc4a13a9ca2630c623cc500e80f7ed6800bb5eb9

closed 2023-02-10 by JiaoXianjun