open-sdr/openwifi #274: Quick Start Tutorial Problem

Baraday3, 2023-02-12:
Hello,

I downloaded openwifi img and I burnt the SD card on Windows 10 with Etcher. Then I coppied boot files of the ADRV9364-Z7020 to base boot directory. I also checkhed rootfs for kernel_modules and interface.new files. But they did not exist. Then, I connected via SSH to my board. I run ./openwifi/setup_once.sh command. After that I run ./wgd.sh and ./fosdem.sh commands. Unfortunatelly, I got error. You can find the output logs of the scripts.

[outputof_fosdem_sh.txt](https://github.com/open-sdr/openwifi/files/10715450/outputof_fosdem_sh.txt)
[outputof_wgd.sh.txt](https://github.com/open-sdr/openwifi/files/10715451/outputof_wgd.sh.txt)

JiaoXianjun, 2023-02-12:
Can you share me output of:
the file name of the downloaded img file
run dmesg on board (after run wgd.sh and fosdem.sh)
run:
./sdrctl dev sdr0 get reg drv_xpu 7
./sdrctl dev sdr0 get reg rx 31
./sdrctl dev sdr0 get reg xpu 63

Baraday3, 2023-02-12:
Hello Jiao, thank you for your fast response.
Name of the downloaded image is openwifi-1.4.0-notter.

[dmesg_output.txt](https://github.com/open-sdr/openwifi/files/10716472/dmesg_output.txt)

root@analog:~/openwifi# ./sdrctl dev sdr0 get reg drv_xpu 7
command failed: No such device (-19)
root@analog:~/openwifi# ./sdrctl dev sdr0 get reg rx 31
command failed: No such device (-19)
root@analog:~/openwifi# ./sdrctl dev sdr0 get reg xpu 63
command failed: No such device (-19)

JiaoXianjun, 2023-02-12:
I think the problem is in wgd.sh outputs:
```
./rf_init_11n.sh: 85: cannot create in_voltage1_gain_control_mode: Permission denied
```
I never see this kind of issue before.

Might be due to the SD card flashing issue (some directory permissions change/reserve during flashing?). Since I never flash SD card in Windows, I don't know what is the root cause.

Maybe you could try to do the SD card flashing in Linux, or try to solve that permission issue on board by command like "chmod", "chown".

closed 2023-02-13 by Baraday3