open-sdr/openwifi #275: Multiple SDR interfaces (SDRPi Hardware)

nrathaus, 2023-02-12:
0. Could you send email to xianjun.jiao@ugent.be to introduce your self?
sent an email

1. Our image is used directly or you build your own image?
Yours

3. What is your own modification?
None

5. Versions: OS, Vivado, openwifi/openwifi-hw repo branch and commit revision
Linux analog 5.10.0-98248-g1bbe32fa5182-dirty #2 SMP PREEMPT Fri Jan 27 14:10:43 CET 2023 armv7l GNU/Linux

7. Board/hardware type
SDRPi

9. WiFi channel number
Not relevant

11. Steps to reproduce the issue, and the related error message, screenshot, etc
dmesg shows multiple SDR interfaces, OpenWiFi doesn't seem to provide the ability to "hook" the other SDR interfaces, only the primary one

12. Describe your debug efforts by Linux native tools, such as tcpdump and "cat /proc/interrupts"

13. Describe your debug efforts by: https://github.com/open-sdr/openwifi/blob/master/doc/README.md#Debug-methods

14. Any other thing we need to know for helping you better?

JiaoXianjun, 2023-02-12:
As far as I know, so far we only have one WiFi interface and one Ethernet interface in our default img.
I don't know your img and your changes. Maybe the best way is to contact the people who made your img.

nrathaus, 2023-02-12:
I used this
https://users.ugent.be/~xjiao/openwifi-1.4.0-notter.img.xz

On the SDRPi board -
https://github.com/HexSDR/sdrpi-fw

The hardware supports two SDRs but only one shows up

So I am guessing the issue is with the default image?

Can you guide me into what changes are needed? I don’t see sdr.ko
supporting something that would allow several interfaces

JiaoXianjun, 2023-02-12:
I don’t understand. We never use more than one WiFi interface and Ethernet interfaces.  

You need to tell me how to reproduce the issue step by step (from power on your board).

nrathaus, 2023-02-12:
I downloaded the image from the web site (the link is previously mentioned)

Wrote it to the SD card, copied the sdrpi drivers to the BOOT folder,
booted the device with the SD card

Reached the prompt via SSH on 192.168.10.122

Ran the openwifi SDR setup sh script

Then the wgd.sh script which brings up the interface

JiaoXianjun, 2023-02-12:
> Then the wgd.sh script which brings up the interface

“Then the wgd.sh script which brings up the interface”. Then? What is the issue while you run wgd.sh?

nrathaus, 2023-02-12:
There is no issue beyond the fact that only sdr0 shows up while there are
secondary SDRs present

Which brings me to the question how can they be brought up? I don’t see
insmod support such a variable

JiaoXianjun, 2023-02-12:
> while there are secondary SDRs present

I don’t understand this “while there are
secondary SDRs present”. How do you know there are secondary SDR present? What is the secondary SDR? We never put more than one WiFi NIC in FPGA, there is only one WiFi NIC in FPGA, so you can see only sdr0, and no sdr1.

nrathaus, 2023-02-12:
Hi

Please look here
https://www.ebay.com/itm/125468127194

Maybe I am not reading the board correctly, 42-45 talks about two channels
of the RF

Doesn’t this mean there are two SDR? Or at least you should be seeing two
interfaces of WiFi and not one?

I am starting to suspect I misunderstood the explanation of the design on
the page

JiaoXianjun, 2023-02-12:
> Hi Please look here https://www.ebay.com/itm/125468127194 Maybe I am not reading the board correctly, 42-45 talks about two channels of the RF Doesn’t this mean there are two SDR? Or at least you should be seeing two interfaces of WiFi and not one? I am starting to suspect I misunderstood the explanation of the design on the page On Sun, 12 Feb 2023 at 20:49, Jiao Xianjun ***@***.***> wrote: while there are secondary SDRs present
> -- Thanks, Noam Rathaus

"Doesn’t this mean there are two SDR? " -- NO.
For example, when you buy a WiFi AP/rounter with 4 or 8 antennas, they all work for a single NIC (WiFi chip) in the AP . Does not necessarily mean there are multiple NICs (WiFi chips). I advise you check our block diagram to understand better: 
https://github.com/open-sdr/openwifi/blob/master/doc/README.md

nrathaus, 2023-02-12:
Thank you for the clarification.

Sorry for the hassle I might have caused

JiaoXianjun, 2023-02-12:
You are welcome.

closed 2023-02-12 by JiaoXianjun