open-sdr/openwifi #277: Does anyone use OpenWiFi for mesh networking success?

zhangmazi123321, 2023-02-19:
When I use OpenWiFi for Batman-Adv networking, I find that nodes cannot discover each other. I couldn't find a problem, so I configured the mesh directly between the nodes as follows
->iw dev sdr0 interface add mesh0 type mp
->iw dev mesh0 set channel 1
->iw dev mesh0 set meshid mymesh
->ifconfig mesh0 hw ether 00:15:16:10:20:66
->ifconfig mesh0 192.168.13.1
->iw dev mesh0 mesh join mymesh
->iw dev mesh_iface station dump

Nodes also cannot discover each other.
The mesh networking configuration using BATMAN-ADV is as follows
->ifconfig sdr0 up
->batctl if add sdr0
->batctl if
->ip link set mtu 1532 dev sdr0
->ifconfig sdr0 down
->iwconfig sdr0 mode ad-hoc essid my-mesh ap 02:12:34:56:78:9A channel 1
->ifconfig sdr0 up
->batctl if add sdr0
->ip link set up dev sdr0
->ip link set up dev bat0

JiaoXianjun, 2023-02-19:
Did you enable/add mesh support to the kernel by kernel configuration and then re-generate new kernel?

We never try this Batman. But I suspect that mesh needs some kernel features. You can search on internet or in the batman community to find out which feature do you need to add into the kernel. Or also ask them which driver and chip feature do they need.  

Openwifi only work at driver and chip level.

zhangmazi123321, 2023-02-19:
Yes, I checked the corresponding option in the kernel and recompiled the kernel, and finally generated the corresponding uImage. This is an uncomplicated configuration, but neighboring nodes are not discoverable to each other. I didn't figure out the problem. I hope that a friend who has made it can guide me.

JiaoXianjun, 2023-02-21:
What you have done to debug the issue?
If I were you, I will try to capture and analyze the packet sent/received in Linux, in openwifi driver and in the air.
You can use tcpdump, wireshark, and also our openwifi driver printing: https://github.com/open-sdr/openwifi/blob/master/doc/README.md#debug-methods

zhangmazi123321, 2023-02-28:
> What you have done to debug the issue? If I were you, I will try to capture and analyze the packet sent/received in Linux, in openwifi driver and in the air. You can use tcpdump, wireshark, and also [...]

Hello, I am having the same problem as in the following link.
https://github.com/open-sdr/openwifi/issues/186
In addition, I want to know if mesh networking with OpenWiFi is a feasible direction, or has anyone successfully made it?

peking428, 2023-03-10:
thanks JiaoXianjun.   openwifi is very strong.  i finished this work that runing batman base on AP-sta/mesh/ad-hoc with this three mode.  debug this work, need some must step and ability of resolve problem. so. v :  peking428

zhangmazi123321, 2023-03-16:
> 

Hello, is the batman protocol you are using based on the three-layer one? There doesn't seem to be mesh mode in OpenWiFi, is this what you get after modifying the driver?

peking428, 2023-03-16:
> > 
> 
> 你好，你用的蝙蝠侠协议是基于三层的吗？OpenWiFi里面好像没有mesh模式，这是你修改驱动后得到的吗？

at MAC layer， not IP layer。 modify driver code is not necessary.

zhangmazi123321, 2023-03-16:
> > > 
> > 
> > 
> > 你好，你用的蝙蝠侠协议是基于三层的吗？OpenWiFi里面好像没有mesh模式，这是你修改驱动后得到的吗？
> 
> at MAC layer， not IP layer。 modify driver code is not necessary.

I see, you are using the BATMAN-ADV protocol. But I still don't understand how you can run in mesh mode without modifying the driver. In addition, have you implemented a three-layer mesh protocol on OpenWiFi?

closed 2023-04-28 by JiaoXianjun

Maxelectronics, 2023-08-30:
> thanks JiaoXianjun. openwifi is very strong. i finished this work that runing batman base on AP-sta/mesh/ad-hoc with this three mode. debug this work, need some must step and ability of resolve problem. so. v : peking428

Can you explain a little bit more how did you do this job please ?

peking428, 2023-09-03:
need modify some kernel file ，.sh，and sdr driver code.

---- Replied Message ----
| From | ***@***.***> |
| Date | 08/30/2023 15:05 |
| To | ***@***.***> |
| Cc | ***@***.***>***@***.***> |
| Subject | Re: [open-sdr/openwifi] Does anyone use OpenWiFi for mesh networking success? (Issue #277) |

thanks JiaoXianjun. openwifi is very strong. i finished this work that runing batman base on AP-sta/mesh/ad-hoc with this three mode. debug this work, need some must step and ability of resolve problem. so. v : peking428

Can you explain a little bit more how did you do this job please ?

Maxelectronics, 2023-10-19:
Which kernel file should be modify please ?

Maxelectronics, 2023-10-20:
> When I use OpenWiFi for Batman-Adv networking, I find that nodes cannot discover each other. I couldn't find a problem, so I configured the mesh directly between the nodes as follows ->iw dev sdr0 [...]

Hello, how did you download and install batman-adv on your plateform ? When I tried, it says that the linux kernel did'nt have linux-headers necessary for the installation ?

Maxelectronics, 2023-11-29:
> need modify some kernel file ，.sh，and sdr driver code.
> […](#)

Which Linux kernel version did you use ? Did you modify something with the command make menuconfig ?