open-sdr/openwifi #280: sdr0 not showing up

mcrabtree1, 2023-02-23:
sdr0 not showing up for openwifi img on [ADRV9364-Z7020 + ADRV1CRR-BOB (LVDS mode)]. Tried to reflash image, tried retriggering udev, tried re-running ./openwifi/setup_once.sh. Cannot find the problem. Below are some outputs 

dmesg output

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.10.0-98248-g1bbe32fa5182-dirty (jxj@jxj-pc) (arm-xilinx-linux-gnueabi-gcc.real (GCC) 10.2.0, GNU ld (GNU Binutils) 2.35.0.20200730) #3 SMP PREEMPT Tue Feb 7 16:53:49 CET 2023
[    0.000000] CPU: ARMv7 Processor [413fc090] revision 0 (ARMv7), cr=18c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Analog Devices ADRV9364-Z7020 (Z7020/AD9364)
[    0.000000] OF: fdt: earlycon: stdout-path /amba@0/uart@E0001000 not found
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 128 MiB at 0x16800000
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000001fffffff]
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000001fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000001fffffff]
[    0.000000] On node 0 totalpages: 131072
[    0.000000]   Normal zone: 1024 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 131072 pages, LIFO batch:31
[    0.000000] percpu: Embedded 15 pages/cpu s29900 r8192 d23348 u61440
[    0.000000] pcpu-alloc: s29900 r8192 d23348 u61440 alloc=15*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 130048
[    0.000000] Kernel command line: console=ttyPS0,115200 root=/dev/mmcblk0p2 rw earlycon rootfstype=ext4 rootwait clk_ignore_unused cpuidle.off=1
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 371420K/524288K available (8192K kernel code, 803K rwdata, 5732K rodata, 1024K init, 151K bss, 21796K reserved, 131072K cma-reserved, 0K highmem)
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=4 to nr_cpu_ids=2.
[    0.000000] 	Trampoline variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=2
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] efuse mapped to (ptrval)
[    0.000000] slcr mapped to (ptrval)
[    0.000000] L2C: platform modifies aux control register: 0x72360000 -> 0x72760000
[    0.000000] L2C: DT/platform modifies aux control register: 0x72360000 -> 0x72760000
[    0.000000] L2C-310 erratum 769419 enabled
[    0.000000] L2C-310 enabling early BRESP for Cortex-A9
[    0.000000] L2C-310 full line of zeros enabled for Cortex-A9
[    0.000000] L2C-310 ID prefetch enabled, offset 1 lines
[    0.000000] L2C-310 dynamic clock gating enabled, standby mode enabled
[    0.000000] L2C-310 cache controller enabled, 8 ways, 512 kB
[    0.000000] L2C-310: CACHE_ID 0x410000c8, AUX_CTRL 0x76760001
[    0.000000] random: get_random_bytes called from start_kernel+0x33c/0x4dc with crng_init=0
[    0.000000] zynq_clock_init: clkc starts at (ptrval)
[    0.000000] Zynq clock init
[    0.000009] sched_clock: 64 bits at 333MHz, resolution 3ns, wraps every 4398046511103ns
[    0.000029] clocksource: arm_global_timer: mask: 0xffffffffffffffff max_cycles: 0x4ce07af025, max_idle_ns: 440795209040 ns
[    0.000056] Switching to timer-based delay loop, resolution 3ns
[    0.000145] clocksource: ttc_clocksource: mask: 0xffff max_cycles: 0xffff, max_idle_ns: 537538477 ns
[    0.000183] timer #0 at (ptrval), irq=25
[    0.000762] Console: colour dummy device 80x30
[    0.000791] Calibrating delay loop (skipped), value calculated using timer frequency.. 666.66 BogoMIPS (lpj=3333333)
[    0.000808] pid_max: default: 32768 minimum: 301
[    0.000981] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.000995] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.001632] CPU: Testing write buffer coherency: ok
[    0.001667] CPU0: Spectre v2: using BPIALL workaround
[    0.001830] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.002332] Setting up static identity map for 0x100000 - 0x100060
[    0.002470] rcu: Hierarchical SRCU implementation.
[    0.002758] smp: Bringing up secondary CPUs ...
[    0.003534] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.003545] CPU1: Spectre v2: using BPIALL workaround
[... 375 lines cut ...]
Reading rules file: /usr/lib/udev/rules.d/80-drivers.rules
Reading rules file: /usr/lib/udev/rules.d/80-ifupdown.rules
Reading rules file: /usr/lib/udev/rules.d/80-libinput-device-groups.rules
Reading rules file: /usr/lib/udev/rules.d/80-net-setup-link.rules
Reading rules file: /usr/lib/udev/rules.d/80-noobs.rules
Reading rules file: /usr/lib/udev/rules.d/80-udisks2.rules
Reading rules file: /usr/lib/udev/rules.d/85-hwclock.rules
Reading rules file: /usr/lib/udev/rules.d/85-regulatory.rules
Reading rules file: /usr/lib/udev/rules.d/90-alsa-restore.rules
Reading rules file: /usr/lib/udev/rules.d/90-console-setup.rules
Reading rules file: /usr/lib/udev/rules.d/90-libiio.rules
Reading rules file: /usr/lib/udev/rules.d/90-libinput-fuzz-override.rules
Reading rules file: /usr/lib/udev/rules.d/90-pi-bluetooth.rules
Reading rules file: /usr/lib/udev/rules.d/90-pipewire-alsa.rules
Reading rules file: /usr/lib/udev/rules.d/95-cd-devices.rules
Reading rules file: /usr/lib/udev/rules.d/95-dm-notify.rules
Reading rules file: /usr/lib/udev/rules.d/96-e2scrub.rules
Reading rules file: /usr/lib/udev/rules.d/97-hid2hci.rules
Reading rules file: /etc/udev/rules.d/99-com.rules
Reading rules file: /usr/lib/udev/rules.d/99-libsane1.rules
Reading rules file: /usr/lib/udev/rules.d/99-systemd.rules
Reading rules file: /etc/udev/rules.d/99-udc.rules
Failed to open device '/sys...': Invalid argument
Unload module index
Unloaded link configuration context.

JiaoXianjun, 2023-02-27:
what about `ifconfig -a`

mcrabtree1, 2023-02-27:
root@analog:~# ifconfig -a
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.10.122  netmask 255.255.255.0  broadcast 192.168.10.255
        inet6 fe80::5c47:1ff8:cd35:da9d  prefixlen 64  scopeid 0x20<link>
        ether 00:05:f7:80:30:7a  txqueuelen 1000  (Ethernet)
        RX packets 46  bytes 5779 (5.6 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 115  bytes 18886 (18.4 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device interrupt 36  base 0xb000  

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 396  bytes 26138 (25.5 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 396  bytes 26138 (25.5 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

root@analog:~#

JiaoXianjun, 2023-02-27:
what is the `dmesg` output after you do all above operations?

mcrabtree1, 2023-02-27:
root@analog:~/openwifi# dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.10.0-98248-g1bbe32fa5182-dirty (jxj@jxj-pc) (arm-xilinx-linux-gnueabi-gcc.real (GCC) 10.2.0, GNU ld (GNU Binutils) 2.35.0.20200730) #3 SMP PREEMPT Tue Feb 7 16:53:49 CET 2023
[    0.000000] CPU: ARMv7 Processor [413fc090] revision 0 (ARMv7), cr=18c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Analog Devices ADRV9364-Z7020 (Z7020/AD9364)
[    0.000000] OF: fdt: earlycon: stdout-path /amba@0/uart@E0001000 not found
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 128 MiB at 0x16800000
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000001fffffff]
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000001fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000001fffffff]
[    0.000000] On node 0 totalpages: 131072
[    0.000000]   Normal zone: 1024 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 131072 pages, LIFO batch:31
[    0.000000] percpu: Embedded 15 pages/cpu s29900 r8192 d23348 u61440
[    0.000000] pcpu-alloc: s29900 r8192 d23348 u61440 alloc=15*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 130048
[    0.000000] Kernel command line: console=ttyPS0,115200 root=/dev/mmcblk0p2 rw earlycon rootfstype=ext4 rootwait clk_ignore_unused cpuidle.off=1
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 371420K/524288K available (8192K kernel code, 803K rwdata, 5732K rodata, 1024K init, 151K bss, 21796K reserved, 131072K cma-reserved, 0K highmem)
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=4 to nr_cpu_ids=2.
[    0.000000] 	Trampoline variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=2
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] efuse mapped to (ptrval)
[    0.000000] slcr mapped to (ptrval)
[    0.000000] L2C: platform modifies aux control register: 0x72360000 -> 0x72760000
[    0.000000] L2C: DT/platform modifies aux control register: 0x72360000 -> 0x72760000
[    0.000000] L2C-310 erratum 769419 enabled
[    0.000000] L2C-310 enabling early BRESP for Cortex-A9
[    0.000000] L2C-310 full line of zeros enabled for Cortex-A9
[    0.000000] L2C-310 ID prefetch enabled, offset 1 lines
[    0.000000] L2C-310 dynamic clock gating enabled, standby mode enabled
[    0.000000] L2C-310 cache controller enabled, 8 ways, 512 kB
[    0.000000] L2C-310: CACHE_ID 0x410000c8, AUX_CTRL 0x76760001
[    0.000000] random: get_random_bytes called from start_kernel+0x33c/0x4dc with crng_init=0
[    0.000000] zynq_clock_init: clkc starts at (ptrval)
[    0.000000] Zynq clock init
[    0.000010] sched_clock: 64 bits at 333MHz, resolution 3ns, wraps every 4398046511103ns
[    0.000029] clocksource: arm_global_timer: mask: 0xffffffffffffffff max_cycles: 0x4ce07af025, max_idle_ns: 440795209040 ns
[    0.000055] Switching to timer-based delay loop, resolution 3ns
[    0.000144] clocksource: ttc_clocksource: mask: 0xffff max_cycles: 0xffff, max_idle_ns: 537538477 ns
[    0.000182] timer #0 at (ptrval), irq=25
[    0.000760] Console: colour dummy device 80x30
[    0.000788] Calibrating delay loop (skipped), value calculated using timer frequency.. 666.66 BogoMIPS (lpj=3333333)
[    0.000805] pid_max: default: 32768 minimum: 301
[    0.000983] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.000998] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.001633] CPU: Testing write buffer coherency: ok
[    0.001668] CPU0: Spectre v2: using BPIALL workaround
[    0.001833] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.002335] Setting up static identity map for 0x100000 - 0x100060
[    0.002475] rcu: Hierarchical SRCU implementation.
[    0.002763] smp: Bringing up secondary CPUs ...
[    0.003543] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.003552] CPU1: Spectre v2: using BPIALL workaround
[    0.003688] smp: Brought up 1 node, 2 CPUs
[    0.003701] SMP: Total of 2 processors activated (1333.33 BogoMIPS).
[    0.003709] CPU: All CPU(s) started in SVC mode.
[... 186 lines cut ...]
[    6.056371] systemd[1]: Mounted Kernel Debug File System.
[    6.095381] systemd[1]: Started Journal Service.
[    6.301765] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
[    6.736615] systemd-journald[1088]: Received client request to flush runtime journal.
[    6.741062] systemd-journald[1088]: File /var/log/journal/9a314adf4f7b4c659ada5035385c9b39/system.journal corrupted or uncleanly shut down, renaming and replacing.
[    9.028232] Registered mathworks_ip class
[    9.091508] mwipcore 43c00000.mwipcore: IRQ index 0 not found
[... cut ...]
[  215.785180] ad9361 spi0.0: ad9361_probe : enter (ad9364)
[  216.098466] ad9361 spi0.0: ad9361_probe : AD936x Rev 0 successfully initialized
[  217.321149] cf_axi_adc 79020000.cf-ad9361-lpc: ADI AIM (10.01.b) at 0x79020000 mapped to 0xd202e27f, probed ADC AD9364 as MASTER
[  217.345286] cf_axi_dds 79024000.cf-ad9361-dds-core-lpc: Analog Devices CF_AXI_DDS_DDS MASTER (9.01.b) at 0x79024000 mapped to 0x6f5b9a0d, probed DDS AD9364
[  227.605164] tx_intf: loading out-of-tree module taints kernel.

JiaoXianjun, 2023-03-17:
We just updated our SD card img to make it more reliable. Could you try to download and try?

mcrabtree1, 2023-03-27:
That worked thank you very much

closed 2023-03-27 by JiaoXianjun