open-sdr/openwifi #281: driver

mkk0719, 2023-02-25:
Hello，I want to use some function of  driver/tx_intf/tx_intf.c in my own user space  function. Are there any methods other than adding system calls?

mkk0719, 2023-02-26:
I want to use reg_write function of the tx_intf.ko so I change the syscall_32.tbl、sys.c、syscalls.h to add the system call of reg_write in my own computer. And then I use the instruction of update driver(./make_all.sh $XILINX_DIR 32). But it didn't work.

mkk0719, 2023-02-27:
I try to add a system call into kernel but I failed. I build a directory 'mychange' below adi-linux and build mychange.c to define the function and wrote 'obj-y = mychang.o' in Makefile below 'mychange'. I also add 'mychange' into Makefile below the adi-linux. I wrote new number '385 i386 mychange sys_mychange' in syscall_32.tbl and I declare the function in adi-linux/include/uapi/asm-generic/unistd.h.  But I can't use this system call in my userspace function. Can you help me?

JiaoXianjun, 2023-02-27:
you need to learn the code related to sdrctl or sysfs:
https://github.com/open-sdr/openwifi/blob/master/driver/sdrctl_intf.c
https://github.com/open-sdr/openwifi/blob/master/driver/sysfs_intf.c
You can also search sdrctl in the issues. Some others have implemented via this userspace kernel communication channel.

mkk0719, 2023-03-01:
Thank you very much, I have implemented this via sysfs.

closed 2023-03-01 by mkk0719