open-sdr/openwifi #282: wget openwifi-low-acc.mp4 timeout

cloud9477, 2023-03-07:
In the update_sdcard.sh, line 141, wget openwifi-low-aac.mp4 no response.

sudo wget -P ./ https://users.ugent.be/~xjiao/openwifi-low-aac.mp4
--2023-03-07 14:49:25--  https://users.ugent.be/~xjiao/openwifi-low-aac.mp4
Resolving users.ugent.be (users.ugent.be)... 157.193.40.145
Connecting to users.ugent.be (users.ugent.be)|157.193.40.145|:443... failed: Connection timed out.
Retrying.

--2023-03-07 14:51:37--  (try: 2)  https://users.ugent.be/~xjiao/openwifi-low-aac.mp4
Connecting to users.ugent.be (users.ugent.be)|157.193.40.145|:443... ^C

JiaoXianjun, 2023-03-07:
Maybe just a temporary ugent server failure.

Anyway, I just fix it by pointing the video file to a github location. Please check my commit few seconds ago.

Tell me it works or not?

Thanks!

cloud9477, 2023-03-07:
The github link works.

closed 2023-03-08 by JiaoXianjun