open-sdr/openwifi #283: Windowing function

peking428, 2023-03-10:
Hello Brother Jiao, 

In our openwifi logic, in the process of moving the time-domain signal to the frequency-domain (Fourier transform) , the time-domain signal is intercepted by a windowing function. What kind of windowing function is used(hanning ? hamming? blackman ? ...)? Can you know the code in which location, we found that the test spectrum leak problem, I looked for a long time did not find, please give us some help~  , thank you!

peking428, 2023-03-10:
[image]

peking428, 2023-03-10:
> 

What kind of windowing function is used(hanning ? hamming? blackman ? ...)? Can you know the code in which location

JiaoXianjun, 2023-03-10:
We never use any windowing function.

peking428, 2023-03-10:
ok. thank you ~

closed 2023-03-10 by peking428