open-sdr/openwifi #285: csi fuzzer recovery

mkk0719, 2023-03-12:
Hello, I am trying to learn csi_fuzzer. I am using one board as an AP and another board as a client. I am using the command "./csi_fuzzer.sh 1 37 0 43" in a terminal on the AP, and using side_info_display.py to receive the data on the client. How can I restore the received CSI data to its original state before using the fuzzer?

By looking at the csi-fuzzer-principle.png in csi_fuzzer.md,
I think the filtering function at the transmitter is: y[n] = x[n] + c1*x[n-1]+c2*x[n-2]
I think that the form of the inverse filter function at the receiver is:
y(0)=x(0)
y(1)=x(1)-c1*y(0) 
y(2)=x(2)-c1*y(1)-c2*y(0) 
... 
y(n)=x(n)-c1*y(n-1)-c2*y(n-2)

x is the data received by the receiver and y is the csi that has been recovered

In my setting:
c1=(37/128)*1j	c2=(43/128)

However, after modifying side_info_display.py based on this function, the actual CSI values in the Python program are changing, but there is no significant change in the shape of the plotted CSI waveform.

Can you give me some advice? Is my inverse filter function wrong?

mkk0719, 2023-03-16:
I modified the function def parse_side_info(side_info, num_eq, CSI_LEN, EQUALIZER_LEN, HEADER_LEN) in side_info_display.py. 
I stored the CSI values obtained and returned by this function in a global variable called side_info_all. Whenever necessary, I use the CSI values from the 2nd last and last rows in side_info_all as the previously obtained y values.
Then, I calculate using the formula mentioned above.

JiaoXianjun, 2023-03-17:
https://dl.acm.org/doi/abs/10.1145/3448300.3468255

https://github.com/open-sdr/openwifi-hw-img/raw/master/doc_repo/Seppe_Dejonckheere_masterproef.pdf

also try to turn the smoothing off: search "smoothing" in https://github.com/open-sdr/openwifi/blob/master/doc/README.md
play the on fpga loop back to control everything: https://github.com/open-sdr/openwifi/blob/master/doc/app_notes/packet-iq-self-loopback-test.md
also try to turn the fft shift off: search FFT "window shift" in above link.

closed 2023-04-07 by JiaoXianjun