open-sdr/openwifi #287: open WiFi communication distance

zhuzhineng, 2023-03-21:
How much communication distance can be increased by using AD-TRXBOOST1-EBZ power amplifier

JiaoXianjun, 2023-03-30:
I haven’t tested it.

zhuzhineng, 2023-03-30:
> I don't know. Maybe you can check with antsdr manufacture.

Well, at present, we have tested antsdr development board, and its communication distance is only about 100 cm. Is there any way to increase its communication distance？

JiaoXianjun, 2023-03-30:
Can you tell more information?
1. Two antsdr? or antsdr to another phone/laptop?
2. channel number you were using
3. 11a/g? Or 11n? (command/traffic you run at both side for test)
4. Our latest img?
5. Antenna type?
6. etc. (More detailed, better).

zhuzhineng, 2023-03-30:
We use two AntSdrs. The antenna uses the original 9dbi antenna, uses 802.11n, the output frequency is 5.2GHZ, and the channel bandwidth is 20MHZ

 

  
 
 
 
Our latest img?
 
Antenna type?

JiaoXianjun, 2023-04-07:
You can try different channel/band.

closed 2023-04-28 by JiaoXianjun