open-sdr/openwifi #294: transmission with the AGC manual mode

danyvak, 2023-04-02:
0. Could you send email to xianjun.jiao@ugent.be to introduce your self?
I think that I've sent around 6 month ago dany.vak@abra.co.il
1. Our image is used directly or you build your own image?
I used your image
2. What is your own modification?
I'm trying to add 802.11b signals to the design
3. Versions: OS, Vivado, openwifi/openwifi-hw repo branch and commit revision
linux ubuntu server 18.04, vivado 2018.3
4. Board/hardware type
zc706
5. WiFi channel number
7
6. Steps to reproduce the issue, and the related error message, screenshot, etc
I have a general question: my design is not stable, and many times I see that I can't receive DSSS signal. I suspect that my problem is with AGC. But once I switch to the AGC manual mode, the design stop transmit any data. I just wanted to know if it's normal behavior?
In general I know how to debug an application, (vivado, ILA, driver printk) but this one totallly puzzles me. 
7. Describe your debug efforts by Linux native tools, such as tcpdump and "cat /proc/interrupts"
I used a tcpdump and sniffer on the different device to verify that
8. Describe your debug efforts by: https://github.com/open-sdr/openwifi/blob/master/doc/README.md#Debug-methods
In general I see printk from the driver that it's try to send a packet, I will try to check is the phy_tx_start is working, but I can't understand a connection between the AGC and a transmitter
9. Any other thing we need to know for helping you better?

JiaoXianjun, 2023-04-03:
Maybe you should check the RSSI and CSMA/CA part to see whether they believe the channel is busy. If the channel is busy, the tx will be stopped. You can also play with the CSMA/CA settings in this page: https://github.com/open-sdr/openwifi/blob/master/doc/app_notes/frequent_trick.md

danyvak, 2023-04-03:
You are 100% right, it started to transmit once I raised a threshold to CCA -1db, and agc in the manual mode. But I don't understand how those things are connected? Can you give me please some reference where to look for the information?

JiaoXianjun, 2023-04-03:
You can check https://github.com/open-sdr/openwifi-hw/blob/master/ip/xpu/src/cca.v and the related modules to see how do we use rssi and ch_idle.

closed 2023-04-07 by JiaoXianjun