open-sdr/openwifi #295: Add support for header files in user space

lng205, 2023-04-05:
Hello.
I'm trying to install a serial port driver on board, and I'm running into a problem showing that there's no kernel header file in user space. 
I've found a similar problem on ADI's engineer zone:
https://ez.analog.com/linux-software-drivers/f/q-a/565120/kuiper-kernel-version-number-causing-module-loading-and-compiling-issues
Maybe the project should consider adding it to the image or modify the prepare_kernel.sh?

JiaoXianjun, 2023-04-07:
The board OS is from raspberry pi, so you can install anything you want via "apt" command on board. Will this way work?

Or, could you come up with a pull request?

lng205, 2023-04-07:
Thank you for your response. The 'apt' command has been useful, but unfortunately, I couldn't find the driver I need on the list. As a student, I'm not very familiar with Linux, but I believe this issue could affect anyone who wants to compile a C file on board.
If my solution proves effective, I will submit a pull request.

lng205, 2023-04-11:
After editing the /kernel_boot/kernel_config file and enabling the driver, I successfully solved my problem by recompiling and substituting the kernel. However, the issue of the missing kernel header files still persists, which could potentially impact other developers as well.

closed 2023-04-11 by lng205