open-sdr/openwifi #297: ap-client-two-sdr

Fei8686, 2023-04-07:
Hi,
Thank you very much for providing this project of openwifi-v1.4.0, it helps a lot during my study. I am experimenting with a zc706_fmcs2 board.The image file I use is “openwifi-1.4.0-notter.img”.

In the app_notes,I noticed that you have proposed a test process related to interconnection with two boards(app_notes/ap-client-two-sdr.md). However, there seems to be a problem with the network.server running on the "client board".

- After the command `service network-manager stop`,I get the following output on the "client board":

`
Failed to stop network-manager.service: Unit network-manager.service not loaded.`

- Then,I continue to execute the commands provided in the ap-client-two-sdr.md.The following information is displayed on the client side:
```
root@analog:~#cd openwifi
root@analog:~/openwifi# ./wgd.sh
usage:
  Script for load (or download+load) different driver and FPGA img without rebooting
  no  argument: Load .ko driver files and FPGA img (if system_top.bit.bin exist) in current dir with test_mode=0.
  1st argument: If it is a NUMBER, it will be assigned to test_mode. Then load everything from current dir.
  1st argument: If it is a string called "remote", it will download driver/FPGA and load everything.
  - 2nd argument (if exist) is the target directory name for downloading and reloading
  - 3rd argument (if exist) is the value for test_mode
  1st argument: neither NUMBER nor "remote" nor a .tar.gz file, it is regarded as a directory and load everything from it.
  - 2nd argument (if exist) is the value for test_mode
  1st argument: a .tar.gz file, it will be unpacked then load from that unpacked directory
  - 2nd argument (if exist) is the value for test_mode

Module                  Size  Used by
mac80211              565248  0
libarc4                16384  1 mac80211
xilinx_dma             28672  0
ad9361_drv            163840  1
cfg80211              319488  1 mac80211
8021q                  24576  0
mwipcore               16384  0
mwipcore_iio_streaming    16384  1 mwipcore
mwipcore_dma_streaming    20480  1 mwipcore
mathworks_ip_common    16384  1 mwipcore
ipv6                  397312  30
TARGET_DIR ./
DOWNLOAD_FLAG 0
test_mode 0
tx_offset_tuning_enable 0
$TARGET_DIR is found!

hostapd: no process found
dhcpd: no process found
sdr0: ERROR while getting interface flags: No such device
rmmod: ERROR: Module sdr is not currently loaded
./load_fpga_img.sh: line 14: warning: command substitution: ignored null byte in input
xlnx,zynq-7000
.//system_top.bit.bin
system_top.bit.bin
+ ifconfig sdr0 down
sdr0: ERROR while getting interface flags: No such device
+ rmmod sdr
rmmod: ERROR: Module sdr is not currently loaded
+ rmmod openofdm_rx
rmmod: ERROR: Module openofdm_rx is not currently loaded
+ rmmod openofdm_tx
rmmod: ERROR: Module openofdm_tx is not currently loaded
+ rmmod rx_intf
rmmod: ERROR: Module rx_intf is not currently loaded
+ rmmod tx_intf
rmmod: ERROR: Module tx_intf is not currently loaded
+ rmmod xpu
rmmod: ERROR: Module xpu is not currently loaded
+ sleep 1
+ '[' -f .//system_top.bit.bin ']'
+ echo 0
+ mkdir -p /lib/firmware
+ cp .//system_top.bit.bin /lib/firmware/ -rf
+ echo system_top.bit.bin
+ '[' xlnx,zynq-7000 '!=' xlnx,zynq-7000 ']'
+ SPI_DEVNAME=spi0.0
+ DDS_DEVNAME=79024000.cf-ad9361-dds-core-lpc
+ ADC_DEVNAME=79020000.cf-ad9361-lpc
+ '[' '!' -d /sys/bus/spi/drivers/ad9361/spi0.0 ']'
+ cd /sys/bus/spi/drivers/ad9361/
+ echo spi0.0
+ echo spi0.0
+ '[' '!' -d /sys/bus/platform/drivers/cf_axi_adc/79020000.cf-ad9361-lpc ']'
+ cd /sys/bus/platform/drivers/cf_axi_adc/
+ echo 79020000.cf-ad9361-lpc
+ echo 79020000.cf-ad9361-lpc
+ set +x
tx_offset_tuning_enable 0
openwifi_ad9361_fir_tx_0MHz_11n.ftr tx_fir_enable 1
Found openwifi_ad9361_fir_tx_0MHz_11n.ftr
+ test -f /sys/bus/iio/devices/iio:device0/in_voltage_rf_bandwidth
+ test -f /sys/bus/iio/devices/iio:device1/in_voltage_rf_bandwidth
+ test -f /sys/bus/iio/devices/iio:device2/in_voltage_rf_bandwidth
+ cd /sys/bus/iio/devices/iio:device2/
+ set +x
FIR Rx: 48,1 Tx: 48,1
1
1
rx0 agc fast_attack
slow_attack
fast_attack
rx1 agc fast_attack
slow_attack
fast_attack
40000000
25215513
40000000
25215414
rssi
112.00 dB
113.75 dB
rx0 gain to 70
72.000000 dB
./rf_init_11n.sh: 101: echo: echo: I/O error
72.000000 dB
rx1 gain to 70
73.000000 dB
./rf_init_11n.sh: 106: echo: echo: I/O error
73.000000 dB
tx0 gain -89dB
-10.000000 dB
-89.000000 dB
tx1 gain 0dB
-10.000000 dB
0.000000 dB
rmmod: ERROR: Module tx_intf is not currently loaded
+ insmod .//tx_intf.ko
tx_intf is loaded!
rmmod: ERROR: Module rx_intf is not currently loaded
+ insmod .//rx_intf.ko
rx_intf is loaded!
rmmod: ERROR: Module openofdm_tx is not currently loaded
+ insmod .//openofdm_tx.ko
openofdm_tx is loaded!
rmmod: ERROR: Module openofdm_rx is not currently loaded
+ insmod .//openofdm_rx.ko
openofdm_rx is loaded!
rmmod: ERROR: Module xpu is not currently loaded
+ insmod .//xpu.ko
xpu is loaded!
rmmod: ERROR: Module sdr is not currently loaded
+ insmod .//sdr.ko test_mode=0
sdr is loaded!
the end
root@analog:~/openwifi# ifconfig sdr0 up
root@analog:~/openwifi# iwlist sdr0 scan
sdr0      Scan completed :
          Cell 01 - Address: 66:55:44:33:22:E0
                    Channel:36
                    Frequency:5.18 GHz (Channel 36)
                    Quality=54/70  Signal level=-56 dBm
                    Encryption key:on
                    ESSID:"openwifi"
                    Bit Rates:6 Mb/s; 9 Mb/s; 12 Mb/s; 18 Mb/s; 24 Mb/s
                              36 Mb/s; 48 Mb/s; 54 Mb/s
                    Bit Rates:63.5 Mb/s
                    Mode:Master
                    Extra:tsf=0000000000000000
                    Extra: Last beacon: 1000ms ago
                    IE: Unknown: 00086F70656E77696669
                    IE: Unknown: 01088C9298A43048606C
                    IE: Unknown: 030124
                    IE: Unknown: 3201FF
                    IE: Unknown: 2D1A0C0000FF00000000000000000000000100000000000000000000
                    IE: Unknown: 3D1624000000000000000000000000000000000000000000
                    IE: Unknown: 7F080000000200000040
                    IE: WPA Version 1
                        Group Cipher : TKIP
                        Pairwise Ciphers (2) : CCMP TKIP
                        Authentication Suites (1) : PSK
                    IE: Unknown: DD180050F2020101010003A4000027A4000042435E0062322F00

root@analog:~/openwifi# wpa_supplicant -i sdr0 -c wpa-openwifi.conf
Successfully initialized wpa_supplicant
nl80211: deinit ifname=sdr0 disabled_11b_rates=0
sdr0: CTRL-EVENT-TERMINATING
```

- This indicates that both AP and client have been successfully started, and the openwifi information of AP can be scanned on the client side, but AP and client cannot be connected.

Could you help me to solve the problem of how to connect with two boards?

Thanks,
The shy

JiaoXianjun, 2023-04-07:
Maybe it is due to that I add password in the hostapd configuration file. You can use this one to replace the hostapd configuration file on board:
https://github.com/open-sdr/openwifi/blob/master/user_space/hostapd-openwifi.conf

Fei8686, 2023-04-11:
I have checked the hostapd configuration file(hostapd-openwifi.conf). But the content of the file on board is the same as what you provided.

May I ask if the reason why the board cannot be associated is that the network-manager.service was not successfully loaded?

 

 https://github.com/open-sdr/openwifi/blob/master/user_space/hostapd-openwifi.conf

Fei8686, 2023-04-11:
I have already solved this problem. I replaced the content in the hostapd-openwifi-11ag.conf file to the code you provided, and the two boards were successfully connected.

Thank you so much!

closed 2023-04-28 by JiaoXianjun