open-sdr/openwifi #301: Stuck while running wgd.sh

song4399, 2023-04-13:
0. Could you send email to xianjun.jiao@ugent.be to introduce your self?

1. Our image is used directly or you build your own image?

2. What is your own modification?

3. Versions: OS, Vivado, openwifi/openwifi-hw repo branch and commit revision

4. Board/hardware type

5. WiFi channel number

6. Steps to reproduce the issue, and the related error message, screenshot, etc

7. Describe your debug efforts by Linux native tools, such as tcpdump and "cat /proc/interrupts"

8. Describe your debug efforts by: https://github.com/open-sdr/openwifi/blob/master/doc/README.md#Debug-methods

9. Any other thing we need to know for helping you better?

song4399, 2023-04-13:
I installed the latest image and burned it to the SD card, and then installed the quick start configuration through the serial port;
First: ssh root@192.168.10.122
Then: cd openwifi
./wgd.sh
After that the serial monitor gets stuck

song4399, 2023-04-13:
U-Boot 2018.01-21439-gd244ce5 (Jul 29 2021 - 16:31:12 +0100) Xilinx Zynq ZC706, Build: jenkins-development-build_uboot-1

Model: Zynq ZC706 Development Board
Board: Xilinx Zynq
Silicon: v3.1
I2C:   ready
DRAM:  ECC disabled 1 GiB
MMC:   sdhci@e0100000: 0 (SD)
SF: Detected s25fl128s_64k with page size 512 Bytes, erase size 128 KiB, total 32 MiB
*** Warning - bad CRC, using default environment

In:    serial@e0001000
Out:   serial@e0001000
Err:   serial@e0001000
Net:   ZYNQ GEM: e000b000, phyaddr 7, interface rgmii-id
eth0: ethernet@e000b000
reading uEnv.txt
407 bytes read in 20 ms (19.5 KiB/s)
Importing environment from SD ...
Hit any key to stop autoboot:  0
Device: sdhci@e0100000
Manufacturer ID: 3
OEM: 5344
Name: SC16G
Tran Speed: 50000000
Rd Block Len: 512
SD version 3.0
High Capacity: Yes
Capacity: 14.8 GiB
Bus Width: 4-bit
Erase Group Size: 512 Bytes
reading uEnv.txt
407 bytes read in 19 ms (20.5 KiB/s)
Loaded environment from uEnv.txt
Importing environment from SD ...
Running uenvcmd ...
Copying Linux from SD to RAM...
reading uImage
5593704 bytes read in 319 ms (16.7 MiB/s)
reading devicetree.dtb
21449 bytes read in 25 ms (836.9 KiB/s)
** Unable to read file uramdisk.image.gz **
## Booting kernel from Legacy Image at 03000000 ...
   Image Name:   Linux-5.10.0-98248-g1bbe32fa5182
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    5593640 Bytes = 5.3 MiB
   Load Address: 00008000
   Entry Point:  00008000
   Verifying Checksum ... OK
## Flattened Device Tree blob at 02a00000
   Booting using the fdt blob at 0x2a00000
   Loading Kernel Image ... OK
   Loading Device Tree to 1fff7000, end 1ffff3c8 ... OK

Starting kernel ...

C:0x000080C0-0x0055DA40->0x010C9800-0x0161F180
Uncompressing Linux... done, booting the kernel.
Booting Linux on physical CPU 0x0
Linux version 5.10.0-98248-g1bbe32fa5182-dirty (jxj@jxj-pc) (arm-xilinx-linux-gnueabi-gcc.real (GCC) 10.2.0, GNU ld (GNU Binutils) 2.35.0.20200730) #6 SMP PREEMPT Tue Mar 14 12:13:23 CET 2023
CPU: ARMv7 Processor [413fc090] revision 0 (ARMv7), cr=18c5387d
CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
OF: fdt: Machine model: Xilinx Zynq ZC706
OF: fdt: earlycon: stdout-path /amba@0/uart@E0001000 not found
Memory policy: Data cache writealloc
cma: Reserved 128 MiB at 0x38000000
Zone ranges:
  Normal   [mem 0x0000000000000000-0x000000002fffffff]
  HighMem  [mem 0x0000000030000000-0x000000003fffffff]
Movable zone start for each node
[... 458 lines cut ...]
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device interrupt 37  base 0xb000

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 184  bytes 18884 (18.4 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 184  bytes 18884 (18.4 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

root@analog:~# cd ~/openwifi
root@analog:~/openwifi# ./wgd.sh
usage:
  Script for load (or download+load) different driver and FPGA img without rebooting
  no  argument: Load .ko driver files and FPGA img (if system_top.bit.bin exist) in current dir with test_mode=0.
  1st argument: If it is a NUMBER, it will be assigned to test_mode. Then load everything from current dir.
  1st argument: If it is a string called "remote", it will download driver/FPGA and load everything.
  - 2nd argument (if exist) is the target directory name for downloading and reloading
  - 3rd argument (if exist) is the value for test_mode
  1st argument: neither NUMBER nor "remote" nor a .tar.gz file, it is regarded as a directory and load everything from it.
  - 2nd argument (if exist) is the value for test_mode
  1st argument: a .tar.gz file, it will be unpacked then load from that unpacked directory
  - 2nd argument (if exist) is the value for test_mode

JiaoXianjun, 2023-04-28:
You should offer more detailed information according to our template.

closed 2023-06-09 by JiaoXianjun