open-sdr/openwifi #303: Ad9361 get DAC data from IP：util_ad9361_dac_upack, But no ofdm signal can be observed by spectrum analyzer

KrisYuFi, 2023-04-18:
0. Could you send email to xianjun.jiao@ugent.be to introduce your self?
    I sent it once.
1. Our image is used directly or you build your own image?
    I built my own image，I used the Linux kernel provided by ADI , But the deconfig file is used using the one you provide. 
2. What is your own modification?
    I used the openofdm_rx , openofdm_tx and an axi_dma0 to transmit my data, opendm_tx modulates data, then IQ data is fed into util_ad9361_dac_upack through a FIFO. I just want to use your 802.11pyh layer. 
   The pictures below shows that the data has been successfully sent to the AD9361IP core input:
[image]
[image]
   In Linux I used your source file build the ad9361_drv.ko, I wrote my own dma transceiver driver, To enable the tx channel I refer to the code in the following part of your sdr.c and only select the data source as dma:
[image]
Finally, I used the rf_init_11n.sh to config the parameter of ad9361 , tx_lo is 2.85Ghz, gain is -20dB.
But no ofdm signal can be observed by spectrum analyzer.

3. Versions: OS, Vivado, openwifi/openwifi-hw repo branch and commit revision
    In Ubuntu16.04, Vivado2019.1 and 2018.3
4. Board/hardware type
   fmcomms2 + zc702

7. Describe your debug efforts by Linux native tools, such as tcpdump and "cat /proc/interrupts"
    I have never encountered this problem. In the case of no-os, it can be used normally after setting the data source. There are few demo operations of AD9361 under Linux on Github, and the most popular one is openwifi. In /sys/bus/iio/devices/iio:device1 , parameters are right.

9. Any other thing we need to know for helping you better? 
    Before I set the data source from DMA I could see monophonic signals on the  spectrum analyzer, after that it was just noise. My thought is, are there need any other configuration after I set the data source? I can turn on wifi normally when using normal openwifi image, so it should not be a hardware problem, right?

JiaoXianjun, 2023-04-18:
https://github.com/open-sdr/openwifi/blob/77071055782bad3582085ea944dde93bbbf2598d/driver/sdr.c#L2224
Please make sure the ad9361 select DMA as data source, not DDS.

KrisYuFi, 2023-04-18:
> https://github.com/open-sdr/openwifi/blob/77071055782bad3582085ea944dde93bbbf2598d/driver/sdr.c#L2224
> Please make sure the ad9361 select DMA as data source, not DDS.

Yes，I select DMA as data source, but     
I can't see ofdm signal on  the Analyzer.

JiaoXianjun, 2023-04-28:
More ILAs are needed to help your debugging.
We can help little, because we do not have your design.

KrisYuFi, 2023-04-29:
Thank you, Dr. Jiao. I found the problem yesterday. Before using your openofdm_tx IP core I need to package a packet into RAM, then openofdm_tx will automatically fetch data, but my packet is too short, and my own DMA can't write too much data when writing to PL side, so the ofdm signal duration is a little short. Waveforms are rarely seen by spectrum analyzer.

Here is my Design:
DMA data ----> FIFO
[image]
FIFO---->RAM:
This is a simple IP core
[image]
RAM --->openofdm_tx:
[image]

In order to make the test simple, I kept ADI's official DAC DMA and used **libiio** to send data under the operating system, and I will change my own DMA later.

I have learned a lot from your openwifi project in the last 9 months. Thank you very much!!!!!!!

JiaoXianjun, 2023-04-29:
Good to know.