open-sdr/openwifi #304: usrp

Windy-with-sky, 2023-04-18:
0. Could you send email to xianjun.jiao@ugent.be to introduce your self?

1. Our image is used directly or you build your own image?

2. What is your own modification?

3. Versions: OS, Vivado, openwifi/openwifi-hw repo branch and commit revision

4. Board/hardware type

5. WiFi channel number

6. Steps to reproduce the issue, and the related error message, screenshot, etc

7. Describe your debug efforts by Linux native tools, such as tcpdump and "cat /proc/interrupts"

8. Describe your debug efforts by: https://github.com/open-sdr/openwifi/blob/master/doc/README.md#Debug-methods

9. Any other thing we need to know for helping you better?

Windy-with-sky, 2023-04-18:
hello! well, I've been studying your openwifi.So I want to konw whether we can make the usrp connected with openwif and make the USRP as the true AP and send the beacon frame, then the smartphoneconnects to the AP and communicates with it?

haochenku, 2023-04-28:
same question, does openWiFi support USRPs? X310 for example

JiaoXianjun, 2023-04-28:
We haven’t ported it to usrp 

It is feasible. 

You can try. 

https://github.com/open-sdr/openwifi#Porting-guide

closed 2023-04-28 by JiaoXianjun