open-sdr/openwifi #305: openwifi-1.4.0-notter.img.xz will be broken after login in with adrv9361z7035

sczh01, 2023-04-19:
As  the title. Please refer my log file.
last block is for old kernel(4.14) that's no problem. 
By the way, the Uboot-stage is also no problem with  openwifi-1.4.0-notter.img.xz.
[z7035_sdr.txt](https://github.com/open-sdr/openwifi/files/11267484/z7035_sdr.txt)

JiaoXianjun, 2023-04-28:
Our side test does not have any issue.

closed 2023-04-28 by JiaoXianjun

sczh01, 2023-05-03:
Hi JiaoXianjun,

Sorry, which Log you want? the log file [z7035_sdr.txt](https://github.com/open-sdr/openwifi/files/11267484/z7035_sdr.txt)
is my capture.
The line 1-2994 is the log of power on, the system won't response when login automatically in soon.
The line 2995-3055 is the log of u-boot only, that's no problem.
The line 3057-last line is the old kernel. the open WIFI can work normally.

By the way, my version of z7035 is REVC, how about you?

Thanks
Tony