open-sdr/openwifi #307: Network interface not brought up with openwifi firmware on ANTSDR_E310

lng205, 2023-04-22:
I recently flashed my ANTSDR E310 device with the newest firmware image (v1.4.0). However, I'm unable to connect to the device via SSH and pinging the device returns "destination host unreachable." When attempting to use the serial port, the setup_once and wgd scripts ran successfully, but the fosdem script failed with a "no sdr0" error. Additionally, running the ifconfig command only shows the lo device.

On booting the device, I receive the following error message:
>[FAILED] Failed to start Raise network interfaces.
>See 'systemctl status networking.service' for details.

The details of this error are as follows:

>● networking.service - Raise network interfaces
>     Loaded: loaded (/lib/systemd/system/networking.service; enabled; vendor preset: enabled)
>     Active: failed (Result: exit-code) since Thu 2023-02-09 11:08:27 GMT; 4min 10s ago
>       Docs: man:interfaces(5)
>    Process: 1963 ExecStart=/sbin/ifup -a --read-environment (code=exited, status=1/FAILURE)
>   Main PID: 1963 (code=exited, status=1/FAILURE)
>        CPU: 339ms
>
>Feb 09 11:08:25 analog systemd[1]: Starting Raise network interfaces...
>Feb 09 11:08:25 analog ifup[2021]: RTNETLINK answers: No such device
>Feb 09 11:08:25 analog ifup[1963]: ifup: failed to bring up eth0
>Feb 09 11:08:27 analog systemd[1]: networking.service: Main process exited, code=exited, status=1/FAILURE
>Feb 09 11:08:27 analog systemd[1]: networking.service: Failed with result 'exit-code'.
>Feb 09 11:08:27 analog systemd[1]: Failed to start Raise network interfaces.

Additionally, running `ping 192.168.10.1` on the board results in the error message "ping: socket: Address family not supported by protocol; ping: connect: Network is unreachable."

I'm unsure of how to address these issues and would appreciate any suggestions or guidance.

lng205, 2023-04-23:
It seems that I'm using E310v2, which is not currently support by this project. I'll turn to MicroPhase and close this issue.

closed 2023-04-23 by lng205

piondeno, 2024-04-03:
Good day to you!
The board that I have is E310V2 and encounter same issue as you.
May I ask you that, had you solve this issue?
Thank you.

lng205, 2024-04-04:
> Good day to you! The board that I have is E310V2 and encounter same issue as you. May I ask you that, had you solve this issue? Thank you.

I'm sorry but I forget how I handled this problem...You may refer to this repo and follow the setup steps:
https://github.com/lng205/AD9361_SDR