open-sdr/openwifi #312: Real-time frequency switching

pagmatt, 2023-05-15:
Hi,

I am trying to tune the channel frequency using the command:
` ./sdrctl dev sdr0 set reg rf 1 5300;./sdrctl dev sdr0 set reg rf 5 5300 `.

However, this causes the client to disconnect, to the best of my understanding since it fails to receive the beacons from the AP. Moreover, when the client eventually reconnects, the frequency falls back to 5.18 GHz, as specified in the hostapd configuration file (see attachments).

Is there any way to switch frequency without dropping the connection ? Or am I simply using the wrong commands?

I am using version 1.4.0 of the module, installed via the official image and without introducing any modification.
The boards are two Xilinx ZC 706, and the link is setup on channel 36, hw_mode=a

The commands I am using are:
` cd openwifi % ./wgd.sh` on the client and ` cd openwifi % ./wgd.sh & ./fosdem.sh` on the AP to setup the link,
`wpa_supplicant -i sdr0 -c wpa-connect.conf &` to connect the client to the AP, and finally
` ./sdrctl dev sdr0 set reg rf 1 5300;./sdrctl dev sdr0 set reg rf 5 5300 ` to switch frequency at the AP/client

I am attaching the configuration files, and the dmesg output at the client.

 
Thanks and regards,
Matteo P.

pagmatt, 2023-05-15:
It seems that switching simultaneously the frequency on both STA and AP the prevents the connection drop. However, iwconfig still shows the original frequency (5.18 GHz):
```
sdr0    IEEE 802.11  ESSID:"openwifi"  
             Mode:Managed  Frequency:5.18 GHz  Access Point: 66:55:44:33:22:80   
             Bit Rate=6 Mb/s   Tx-Power=20 dBm
```
despite dmesg logs seem to indicate that the switch (to 3.5 GHz in this case) was completed successfully:
```
[ 1018.260242] sdr,sdr ad9361_tx_calibration 3500MHz tx_quad_cal duration unknown us
[ 1018.260256] sdr,sdr clk_set_rate TX_RFPLL 3500MHz done
[ 1018.267123] sdr,sdr recv set cmd reg cat 1 addr 00000014 val 00000dac idx 5
[ 1018.268727] sdr,sdr openwifi_rf_rx_update_after_tuning 3500MHz rssi_correction 153 fpga_lbt_th 182(-62dBm) auto 182 static 306 receiver th 136(-85dBm)
[ 1018.268737] sdr,sdr clk_set_rate RX_RFPLL 3500MHz done
```

Where can I double check the actual operating frequency of the link?

JiaoXianjun, 2023-05-23:
You can always look for dmesg printing from this function openwifi_rf_rx_update_after_tuning in sdr.c. Each time we change the frequency, we call this function.

You can check ad9361 status, such as out_altvoltage0_RX_LO_frequency, like in this page: https://wiki.analog.com/resources/tools-software/linux-drivers/iio-transceiver/ad9361

closed 2023-06-05 by JiaoXianjun