open-sdr/openwifi #313: how to build openwifi sd image?

yangnsd, 2023-06-06:
Hello,Xianjun Jiao

I follow this link https://github.com/open-sdr/openwifi/blob/master/doc/img_build_instruction/kuiper.md Update_ Sdcard.sh.

Firstly,I download image_2022-08-04-ADI-Kuiper-full.zip,Step by step execution,When I execute Update_ sdcard.sh script, I didn't know which parameters to fill in and which directory the parameters pointed to. I tried to execute the following code, and the terminal would display some options. I didn't know how to check these options, so I pressed Enter all the time. Finally,SD is inserted into the board and I find that it cannot pass SSH root@192.168.10.122 Log in to the board

cd openwifi/user_space
./update_sdcard.sh /home/yjs/Downloads/openwifi-hw/ /opt/Xilinx/ /media/yjs/

So,I would like to know how many parameters need to be filled in at the back of the board, and the meaning of each parameter.

JiaoXianjun, 2023-06-09:
As we said in that script:
```
echo "You must enter exactly 3 arguments: \$OPENWIFI_HW_IMG_DIR \$XILINX_DIR \$SDCARD_DIR"
```
The first argument should be openwifi-hw-img, not openwifi-hw.

You can download/clone openwifi-hw-img here: https://github.com/open-sdr/openwifi-hw-img

yangnsd, 2023-06-12:
Okay, thank you very much!

closed 2023-06-12 by JiaoXianjun