open-sdr/openwifi #314: Openwifi Linux Kernel failed insmod

mliuvanteon, 2023-06-09:
This is what I did:
1. Install WSL on windows 10.
2. Install Ubuntu 18.04.5 on WSL.
3. git clone https://github.com/open-sdr/openwifi

Prepare Analog Devices Kernel

5. cd openwifi/user_space
6. ./prepare_kernel.sh $XILINX_DIR 32

Compile the latest openwifi driver

8. cd openwifi/driver;
9. ./make_all.sh $XILINX_DIR 32

Boot the zed-fmcs2 board with the newly build uImage

11. copy adi-linux/arch/arm/boot/uImage to SD card
12. Boot the zed-fmcs2 dev board

Install the newly built drivers

14. scp the .ko module files to the zed-fmcs2 board

Both insmod and depmod failed

16. depmod failed as follows:
depmod: ERROR: could not open directory /lib/modules/5.10.0-98248-g1bbe32fa5182-dirty: No such file or directory
depmod: FATAL: could not search modules: No such file or directory
17. insmod failed as follows:
insmod: ERROR: could not insert module sdr.ko: Invalid module format

The linux kernel image and the openwifi drivers were built using the same linux source yet insmod still failed.
We want to modify the openwifi drivers which is impossible without insmod working.

JiaoXianjun, 2023-06-09:
Sorry that we never try Ubuntu 18.04.5 on WSL in Win10, so it will be very difficult for us to help here.
I suggest you use a real Ubuntu machine, or use Ubuntu in VMware or Virtualbox.

mliuvanteon, 2023-06-09:
Thank you Xianjun for the quick response.

chenmo234, 2024-06-03:
I also need to modify sdr.c and perform the same operations as you. Can I discuss this with you? Here is my contact information: c2385405729.