open-sdr/openwifi #317: Start with Openwifi

Maxelectronics, 2023-06-21:
I am currently trying to implement Openwifi on an ADRV9361-Z7035-BOB board and I would like to know where exactly we should copy the BOOT.bin and devicetree.tdb files? One more thing, when we want to connect the board to our PC, should the Ethernet cable from our PC be connected to the board? Or should we leave it connected to the router and then change its address?

JiaoXianjun, 2023-06-21:
Can you fill in the template? We need more information to understand the circumstances.
Normally you don't need to do manual copy of .bin and .dtb. You can follow this: https://github.com/open-sdr/openwifi#update-fpga

Normally we connect the board to PC directly via Ethernet cable. But this is standard ip/ethernet thing, you can decide your way.

Maxelectronics, 2023-06-21:
I don't see where it says to build the FPGA. I followed the instructions, but I encounter an error when running wgd.sh.

cd openwifi
root@analog:~/openwifi# ./wgd.sh
usage:
  Script for load (or download+load) different driver and FPGA img without rebog
  no  argument: Load .ko driver files and FPGA img (if system_top.bit.bin exist.
  1st argument: If it is a NUMBER, it will be assigned to test_mode. Then load .
  1st argument: If it is a string called "remote", it will download driver/FPGA.
  - 2nd argument (if exist) is the target directory name for downloading and reg
  - 3rd argument (if exist) is the value for test_mode
  1st argument: neither NUMBER nor "remote" nor a .tar.gz file, it is regarded .
  - 2nd argument (if exist) is the value for test_mode
  1st argument: a .tar.gz file, it will be unpacked then load from that unpackey
  - 2nd argument (if exist) is the value for test_mode
 
insmod: ERROR: could not load module ad9361_drv.ko: No such file or directory
insmod: ERROR: could not load module xilinx_dma.ko: No such file or directory
modprobe: FATAL: Module mac80211 not found in directory /lib/modules/5.10.0-982y
Module                  Size  Used by
TARGET_DIR ./
DOWNLOAD_FLAG 0
test_mode 0
tx_offset_tuning_enable 0
$TARGET_DIR is found!
                                                                                
hostapd: no process found                                                       
dhcpd: no process found                                                         
sdr0: ERROR while getting interface flags: No such device                       
rmmod: ERROR: Module sdr is not currently loaded                                
.//system_top.bit.bin not found. Skip reloading FPGA.                           
tx_offset_tuning_enable 0                                                       
openwifi_ad9361_fir_tx_0MHz_11n.ftr tx_fir_enable 1                             
Found openwifi_ad9361_fir_tx_0MHz_11n.ftr                                       
+ test -f /sys/bus/iio/devices/iio:device0/in_voltage_rf_bandwidth              
+ test -f /sys/bus/iio/devices/iio:device1/in_voltage_rf_bandwidth              
+ test -f /sys/bus/iio/devices/iio:device2/in_voltage_rf_bandwidth              
+ test -f /sys/bus/iio/devices/iio:device3/in_voltage_rf_bandwidth              
+ test -f /sys/bus/iio/devices/iio:device4/in_voltage_rf_bandwidth              
+ echo Can not find in_voltage_rf_bandwidth!                                    
Can not find in_voltage_rf_bandwidth!                                           
+ echo Check log to make sure ad9361 driver is loaded!                          
Check log to make sure ad9361 driver is loaded!                                 
+ exit 1                                                                        
rmmod: ERROR: Module tx_intf is not currently loaded                            
+ insmod .//tx_intf.ko                                                          
insmod: ERROR: could not load module .//tx_intf.ko: No such file or directory   
tx_intf is not loaded!

Maxelectronics, 2023-06-22:
I need to update my FPGA before this steps ? 

raspi-config --expand-rootfs (Only needed when your SD card > 16GB. Run and reboot)
./openwifi/setup_once.sh (Reboot the board. Only need to run once for new board)
cd openwifi
./wgd.sh
./fosdem.sh
(Use "./wgd.sh 1" to enable experimental AMPDU aggregation on top of 11n)
(Use "./fosdem-11ag.sh" to force 11a/g mode)

JiaoXianjun, 2023-06-22:
1. Please fill the issue template.
2. Please describe in details about the exact operation and its output step by step to reproduce this issue. Description should start from you download our SD card img file, and how do you flash it to SD card with all info (OS version, etc, as listed in our issue template).