open-sdr/openwifi #318: Many error reports when running ./wgd.sh and ./fosdem.sh during Demo test

wuyuehua19, 2023-06-22:
0. Could you send email to xianjun.jiao@ugent.be to introduce your self?
       Sent
1. Our image is used directly or you build your own image?
       I used directly image from the server "openwifi-1.4.0-notter.img"
2. What is your own modification?
       No modification
3. Versions: OS, Vivado, openwifi/openwifi-hw repo branch and commit revision
       ubantu18.04 to flash SD card
4. Board/hardware type
        **neptunesdr**, (zynq7020+ad9361)
5. WiFi channel number
        Not reach this step so far.
6. Steps to reproduce the issue, and the related error message, screenshot, etc
        I followed the quick start:
          a. flash SD card with image openwifi-1.4.0-notter.img.
          b. /openwifi/setup_once.sh, then reboot the board
          c. cd openwifi   ->   ./wgd.sh    -> ./fosdem.sh
          d. then there are many error reported as below
              
[error_wgd_fosdem.txt](https://github.com/open-sdr/openwifi/files/11834766/error_wgd_fosdem.txt)
      
[image]
[image]

JiaoXianjun, 2023-06-22:
Maybe your hardware is different. Can you show me your hardware photo? Especially the version?

wuyuehua19, 2023-06-23:
FYI, it's NeptuneSDR, K210
[image]
[image]

wuyuehua19, 2023-06-23:
Dump dmesg info with cmd "./wgd.sh", it report "Invalid bitstream, could not find a sync word. Bitstream must be a byte swapped .bin file", not sure if related to my issue.
[image]

JiaoXianjun, 2023-06-23:
Your hardware is the newer version compared with ours.
You can ask the seller/author of neptune sdr for new img.
We will also check and support soon.

wuyuehua19, 2023-06-23:
Thanks for your response. Will communicate with seller, and hope it's supported soon!

JiaoXianjun, 2023-08-01:
You pick the new FPGA img from: https://github.com/open-sdr/openwifi-hw-img/tree/neptunesdr_new
The changes are: https://github.com/open-sdr/openwifi-hw/tree/neptunesdr_new

mrsaad381, 2024-08-24:
Hi Dear concerned
i am facing some issue similar to above one  i have sampe board Neptune SDR B210 PLUS
i  burn open wifi image file  and i can access board on my pc via ssh root@192.168.10.122
but when i run first file from openwifi directory
./setup_once.sh
it give me errors in make file
also if i run further commands
./wgd.sh
it all time give me blinking of cursor nothing happend
and again ./fosdem.sh
it not detect sdr0 board or same error mention in above issue please guide me with these step 
Thanks in advance 
Here;s attached screenshot of error
[image]

mrsaad381, 2024-08-27:
[image]
pls see when i run wgd.sh its stuck on this line  and after sometime its shows your connection is reset and i disconnect to neptunesdr 
i have same board which mention in above chat 
and my sequence is already mention above
burn your notter 14.0 img to sd card and do all steps  mention in  redme .md to copy paste neptunesdr files to Boot main driectory and  if i am deleting interfaces file so its not able to connect board via ehternet

mrsaad381, 2024-08-27:
@JiaoXianjun