open-sdr/openwifi #319: packet rate modification

nandongnan, 2023-06-26:
Hello! I'm exploring the possibility of using openwifi for sensing, and I found that about 7 packets can be received per second, which seems to be insufficient for sensing. Modifying the MCS did not  work either. I didn't find a way to change the packet rate. Is there a way to modify the packet rate here?Thanks!

JiaoXianjun, 2023-07-11:
how do you transmit the packet?
did you try: https://github.com/open-sdr/openwifi/blob/master/doc/app_notes/inject_80211.md

nandongnan, 2023-08-18:
Yes, I use the inject function to inject packets, and another device running openwifi can monitor the packets. I also made a preliminary attempt under the CSI radar function.Whether there is a device connected to the openwifi, according to the timestamp, it can be seen that the packet rate is about 0.1 second per packet. I would like to ask how to change the packet rate under the CSI radar function?

JiaoXianjun, 2023-08-18:
You could consider methods other than our current examples, if you fully understand the full flow from Linux user space to the FPGA. Possibilities:

1. Do some FPGA sequencer logic to change the FIR settings in FPGA ( could be triggered by your newly defined registers). This should be the fastest way. 

2. Consider access the CSI fuzzer/FIR register via C code as in this example: https://github.com/open-sdr/openwifi/tree/master/user_space/fast_reg_log . this could be faster than our current CSI fuzzer example.

nandongnan, 2023-08-24:
Thank you for your suggestion. When I use a device running openwifi to inject packets through the inject function, and the other one is in monitor mode to capture CSI, I found that when the interval between sending packets is 1 ms or 10 ms or less,  Openwifi can't capture every packet accurately through the curved timestamp line , even if you use side_ch_ctl g0, there will also be packet loss. Is this normal?

JiaoXianjun, 2023-08-25:
Yes. It is possible. Side channel work’s asynchronously with baseband.