open-sdr/openwifi #323: Openwifi in ADRV9361z7035 shut down

frigo125, 2023-07-10:
0. Could you send email to xianjun.jiao@ugent.be to introduce your self?
sent
1. Our image is used directly or you build your own image?
your image
2. What is your own modification?
none
3. Versions: OS, Vivado, openwifi/openwifi-hw repo branch and commit revision
Ubuntu 18.04, Vivado 2021.1, master branch,
4. Board/hardware type
adrv9361z7035
5. WiFi channel number
1
6. Steps to reproduce the issue, and the related error message, screenshot, etc

Hello DR.JIAO, 
We used adrv9361z7035 to implement Openwifi, according to your Quick start, we flash the image to a SD card, and copy  the files in BOOT/openwifi/adrv9361z7035 to BOOT, and insert my SD card into Board and power on, Linux was started normally. when Linux startup is finish. we can login and see root@analog:~# , about 6 seconds, the system shut down, and we can't do anything , (for example, cd openwifi ), and green/yellow leds are still illume. 
[image]

JiaoXianjun, 2023-07-11:
1. Could save the dmesg output right after its booting? (before freezing)
2. Could you make sure use our latest img? (old img's devicetree could cause this issue)
3. Could you try different SD card flashing tool under different OS?

frigo125, 2023-07-11:
1，I save the dmesg output before freezing.
root@analog:~# dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.10.0-98248-g1bbe32fa5182-dirty (jxj@jxj-pc) (arm-xilinx-linux-gnueabi-gcc.real (GCC) 10.2.0, GNU ld (GNU Binutils) 2.35.0.20200730) #6 SMP PREEMPT Tue Mar 14 12:13:23 CET 2023
[    0.000000] CPU: ARMv7 Processor [413fc090] revision 0 (ARMv7), cr=18c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Analog Devices ADRV9361-Z7035 (Z7035/AD9361)
[    0.000000] OF: fdt: earlycon: stdout-path /amba@0/uart@E0001000 not found
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 128 MiB at 0x16800000
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000001fffffff]
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000001fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000001fffffff]
[    0.000000] On node 0 totalpages: 131072
[    0.000000]   Normal zone: 1024 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 131072 pages, LIFO batch:31
[    0.000000] percpu: Embedded 15 pages/cpu s29900 r8192 d23348 u61440
[    0.000000] pcpu-alloc: s29900 r8192 d23348 u61440 alloc=15*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 130048
[    0.000000] Kernel command line: console=ttyPS0,115200 root=/dev/mmcblk0p2 rw earlycon rootfstype=ext4 rootwait clk_ignore_unused cpuidle.off=1
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 371432K/524288K available (8192K kernel code, 803K rwdata, 5732K rodata, 1024K init, 151K bss, 21784K reserved, 131072K cma-reserved, 0K highmem)
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=4 to nr_cpu_ids=2.
[    0.000000]  Trampoline variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=2
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] efuse mapped to (ptrval)
[    0.000000] slcr mapped to (ptrval)
[    0.000000] L2C: platform modifies aux control register: 0x72360000 -> 0x72760000
[    0.000000] L2C: DT/platform modifies aux control register: 0x72360000 -> 0x72760000
[    0.000000] L2C-310 erratum 769419 enabled
[    0.000000] L2C-310 enabling early BRESP for Cortex-A9
[    0.000000] L2C-310 full line of zeros enabled for Cortex-A9
[    0.000000] L2C-310 ID prefetch enabled, offset 1 lines
[    0.000000] L2C-310 dynamic clock gating enabled, standby mode enabled
[    0.000000] L2C-310 cache controller enabled, 8 ways, 512 kB
[    0.000000] L2C-310: CACHE_ID 0x410000c8, AUX_CTRL 0x76760001
[    0.000000] random: get_random_bytes called from start_kernel+0x33c/0x4dc with crng_init=0
[    0.000000] zynq_clock_init: clkc starts at (ptrval)
[    0.000000] Zynq clock init
[    0.000009] sched_clock: 64 bits at 333MHz, resolution 3ns, wraps every 4398046511103ns
[    0.000029] clocksource: arm_global_timer: mask: 0xffffffffffffffff max_cycles: 0x4ce07af025, max_idle_ns: 440795209040 ns
[    0.000054] Switching to timer-based delay loop, resolution 3ns
[    0.000145] clocksource: ttc_clocksource: mask: 0xffff max_cycles: 0xffff, max_idle_ns: 537538477 ns
[    0.000182] timer #0 at (ptrval), irq=25
[    0.000728] Console: colour dummy device 80x30
[    0.000757] Calibrating delay loop (skipped), value calculated using timer frequency.. 666.66 BogoMIPS (lpj=3333333)
[    0.000774] pid_max: default: 32768 minimum: 301
[    0.000951] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.000964] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.001601] CPU: Testing write buffer coherency: ok
[    0.001636] CPU0: Spectre v2: using BPIALL workaround
[    0.001801] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.002306] Setting up static identity map for 0x100000 - 0x100060
[    0.002446] rcu: Hierarchical SRCU implementation.
[    0.002734] smp: Bringing up secondary CPUs ...
[    0.003511] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.003521] CPU1: Spectre v2: using BPIALL workaround
[    0.003660] smp: Brought up 1 node, 2 CPUs
[    0.003673] SMP: Total of 2 processors activated (1333.33 BogoMIPS).
[... 188 lines cut ...]
[    6.366015] systemd[1]: Started Journal Service.
[    6.442066] random: crng init done
[    6.444168] random: 7 urandom warning(s) missed due to ratelimiting
[    6.842908] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
[    7.075499] systemd-journald[1092]: Received client request to flush runtime journal.
[    7.103487] systemd-journald[1092]: File /var/log/journal/9a314adf4f7b4c659ada5035385c9b39/system.journal corrupted or uncleanly shut down, renaming and replacing.
[    9.321957] Registered mathworks_ip class
[    9.409007] mwipcore 43c00000.mwipcore: IRQ index 0 not found
[... cut ...]
[   12.965396] cfg80211: loaded regulatory.db is malformed or signature is missing/invalid
[   15.058726] systemd-journald[1092]: File /var/log/journal/9a314adf4f7b4c659ada5035385c9b39/user-1000.journal corrupted or uncleanly shut down, renaming and replacing.
[   30.222757] PM: suspend entry (deep)

2, I download the image file from open-sdr/opemwifi yesterday, I think it's the latest image file.

3, No, but  I use the new SD card, and the file for other board is OK.(e310V2, zedboard+fmcomms3).

JiaoXianjun, 2023-07-11:
What is your hardware revision: https://wiki.analog.com/resources/eval/user-guides/adrv9361-z7035/revision-history
What is your carrier board:

[FMC Carrier (PZSDRCC-FMC)](https://wiki.analog.com/resources/eval/user-guides/pzsdr/carriers/fmc)
[Breakout Carrier (PZSDRCC-BRK)](https://wiki.analog.com/resources/eval/user-guides/pzsdr/carriers/brk)
[PCIe Carrier (PZSDRCC-PCIE)](https://wiki.analog.com/resources/eval/user-guides/pzsdr/carriers/pcie)
[PackRF Carrier (PZSDRCC-PackRF)](https://wiki.analog.com/resources/eval/user-guides/pzsdr/carriers/packrf)

frigo125, 2023-07-11:
My hardware is Revision E, 
My carrier board is  Breakout Carrier (PZSDRCC-BRK).

Thank you .

frigo125, 2023-07-19:
Hello DR.JIAO,
Can you help me to review this question? 
The last question is my device (adrv9361-z7035) is freezing after booting.
However, I did not do anything, just reboot the device.  but this time, it's work. I dont know what's happened. 
Could it be a hardware problem?

snailxgh, 2023-07-21:
I have the same problem,
My hardware is Revision FAILED
My carrier board is FMC Carrier (PZSDRCC-FMC)

Did you solve it ???

snailxgh, 2023-08-18:
> 1. Could save the dmesg output right after its booting? (before freezing)
> 
>     2. Could you make sure use our latest img? (old img's devicetree could cause this issue)
> 
>     3. Could you try different SD card flashing tool under different OS?

I replaced the device tre and system_top.xsa file, and it doesn't shutdown

JiaoXianjun, 2023-08-18:
We don’t have this issue on our hardwares. So it is difficult for us to know. 

Sometimes, change the SD card ( size, brand, type) and the flashing tool (windows, Linux) help.

closed 2023-09-10 by JiaoXianjun