open-sdr/openwifi #327: Compiling the source code does not generate uImage

snailxgh, 2023-07-17:
I compile openwifi source code.like this:
1. clone source
git clone --recursive https://github.com/open-sdr/openwifi-hw.git
git clone --recursive https://github.com/open-sdr/openwifi-hw-img.git

2. Compiling
cd openwifi/user_space
./update_sdcard.sh ~/workstation/openwifi-hw-img /tools/Xilinx/ /media/devops/

3. Generate Image
Generate Image and openwifi directory in BOOT partition, but I need uImage. My board is  adrv9361z7035

4. Startup Information

U-Boot 2018.01-21439-gd244ce5 (Jul 29 2021 - 16:33:01 +0100), Build: jenkins-development-build_uboot-1

Model: Zynq Zed Development Board
Board: Xilinx Zynq
Silicon: v3.1
DRAM:  ECC disabled 512 MiB
MMC:   sdhci@e0100000: 0 (SD)
SF: Detected n25q256a with page size 256 Bytes, erase size 4 KiB, total 32 MiB
In:    serial@e0001000
Out:   serial@e0001000
Err:   serial@e0001000
Net:   ZYNQ GEM: e000b000, phyaddr 0, interface rgmii-id
eth0: ethernet@e000b000
Hit any key to stop autoboot:  0 
Device: sdhci@e0100000
Manufacturer ID: 3
OEM: 5344
Name: SC32G 
Tran Speed: 50000000
Rd Block Len: 512
SD version 3.0
High Capacity: Yes
Capacity: 29.7 GiB
Bus Width: 4-bit
Erase Group Size: 512 Bytes
** Unable to read file uEnv.txt **
Copying Linux from SD to RAM...
** Unable to read file uImage **
** Unable to read file uramdisk.image.gz **
Wrong Image Format for bootm command
ERROR: can't get kernel image!
Zynq>

JiaoXianjun, 2023-08-14:
Try this:

https://github.com/open-sdr/openwifi/issues/334#issuecomment-1676886785

closed 2023-09-10 by JiaoXianjun

AlexeyIgnatev, 2026-02-17:
Hello, have you found a solution for this problem? Provided solution does not work because there is no ssh access to the board