open-sdr/openwifi #329: Update the uImage file

8663, 2023-07-20:
Hello, Dr. Jiao:
I am trying to modify the contents of the adi-linux/net/mac80211 file.
We compile git command line comments in prepare_kernel.sh to generate uImage file. And replace it with the SD card boot partition.
[image]
Then update FPGA and./make_all.sh.
Finally dmesg, there is no output we want.It seems that the changes we made were not updated to the uImage file.
How to compile and generate uImage file after changing adi-linux/net/mac80211 file?

HavingaThijs, 2023-07-26:
In the image you show, your git commands are commented out (there is a '#' in front of the lines), so they will not be executed. If you uncomment them (remove the '#') and afterwards when replacing the `uImage` file in the boot partition of the SD card, your changes should be applied.

JiaoXianjun, 2023-08-14:
Try this:

https://github.com/open-sdr/openwifi/issues/334#issuecomment-1676886785