open-sdr/openwifi #338: Timestamp

mkk0719, 2023-09-08:
Hi
I have encountered a problem about timestamp. I use one openwifi(antsdr E310) to serve as AP，another openwifi(antsdr E310) to serve as client. I write a simple socket communication between AP and client to communicate some information and at the same time, I get CSI from the client board. 
I found that the timestamp showed in side_info_display.py result is the tsf time of CSI. 
[image]
And I want to align one CSI to the corresponding packet.
I use` tcpdump -i sdr0` to sniffer the packet. And I saw that below. The ip address of AP is 192.168.13.1, the ip address of client is 192.168.13.12.
[image]

The time before IP is the system time because when I running 'date' in the shell of ubuntu, the time is the same.
Could you please help me to explain how to align one CSI to the corresponding packet?

mkk0719, 2023-09-13:
Thank you for your help. I have look through sdr.c. I found that after receiving a packet from the air, FPGA will raise interrupt (if the frame filtering rule allows) to Linux, then the function openwifi_rx_interrupt() of openwifi driver (sdr.c) will be triggered. In that function, ieee80211_rx_irqsafe() API is used to give the packet and related information (timestamp, rssi, etc) to upper layer. 

First, I tried to change the code in openwifi_rx_interrupt() to set the rx_status.mactime the same as the system time.
[image]
But after I compile the driver and try to get csi in monitor mode, the timestamp which printed by side_info_display.py is different from the linux system time. I think my change have not worked.

Then, I have tried to change code in function openwifi_start(struct ieee80211_hw *dev) to change the initial tsf the same as linux system time.
[image]
But I have found that, when I compile the driver and get csi in monitor mode, the timestamp printed by side_info_display.py is different from linux system time.
I think there maybe something I don't understand well when the tsf is given to upper layer. Cause when I set the tsf_time in openwifi_start(), I printk in openwifi_rx_interrupt() to see the rx_status.mactime, mactime is the same as linux system.

Could you please give me some help?

JiaoXianjun, 2023-09-13:
What you did is not what I said in my reply. You need to try what I said.

JiaoXianjun, 2023-09-21:
If you capture packet via:
```
tcpdump -i <interface> -s 65535 -w <file>
```
Then open the file by wireshark, you will find the TSF time of each packet.

mkk0719, 2023-09-24:
I have tried what you said, and I have found the tsf time in wireshark. I'm still trying to align the tsf time and csi information of the same packet.

JiaoXianjun, 2023-09-26:
> I have tried what you said, and I have found the tsf time in wireshark. I'm still trying to align the tsf time and csi information of the same packet.

https://github.com/open-sdr/openwifi/discussions/344