open-sdr/openwifi #339: IQ self loop-back Data Retreival

vishnulatheesh001, 2023-09-15:
Hii @JiaoXianjun ,

First of all your work you put here is very helpfull and great. I was trying to modify something for my testing purpose I highly appreciate any help from you or your team.
I used Open-wifi-notter image and I am able to establish the connection on my Zed Board. I was testing out [IQ self loop-back ](https://github.com/open-sdr/openwifi/blob/master/doc/app_notes/packet-iq-self-loopback-test.md#IQ-self-loopback-quick-start) and I was wondering if I could transmit either some file or message and receive the same. In the current example I am not sure where to modify or which data is being injected. 
To be specific, I want to do Self loop back on my Zed board and transmit some known data and receive the same in receiver side, I also want to know if the example is capable to do the same. If I missed out any example that already exists please point out.

Thanks in advance.
Regards Vishnu

A-T-Kristensen, 2023-09-15:
Hi,

An example of a file with arbitrary IQ values is given here <https://github.com/open-sdr/openwifi/tree/master/user_space/arbitrary_iq_gen>. For transmitting this, see <https://github.com/open-sdr/openwifi/blob/master/user_space/tx_intf_iq_data_to_sysfs.sh>. The documentation is very brief on it, only a little is written here <https://github.com/open-sdr/openwifi/blob/master/doc/app_notes/frequent_trick.md#Arbitrary-Tx-IQ-sample>.

JiaoXianjun, 2023-09-15:
> Hii @JiaoXianjun ,
>
> First of all your work you put here is very helpfull and great. I was trying to modify something for my testing purpose I highly appreciate any help from you or your team. I used [...]

You hit the right place -- that app note is what you need.
You can define the content of your packet in https://github.com/open-sdr/openwifi/blob/master/user_space/inject_80211/inject_80211.c and check the content of the looped back packet by the tcpdump.

JiaoXianjun, 2023-09-15:
> Hi,
>
> An example of a file with arbitrary IQ values is given here https://github.com/open-sdr/openwifi/tree/master/user_space/arbitrary_iq_gen. For transmitting this, see [...]

This is for iq sample injection. If I remember correctly it supports maximum 512 samples. Maybe not that useful.

vishnulatheesh001, 2023-09-22:
Hii @JiaoXianjun ,
 Thank you for your instant reply.

> You can define the content of your packet in https://github.com/open-sdr/openwifi/blob/master/user_space/inject_80211/inject_80211.c and check the content of the looped back packet by the tcpdump.

I have tried this but I am not sure where the data is being printed using the command
`tcpdump -i sdr0`

Please point out where to modify in **inject_80211.c** to generate customised data because in the current code as it was I believe **gen_rand_str** function is generating data but I am not sure. Please guide me.

Thanks in advance