open-sdr/openwifi #342: CSI capture condition

mkk0719, 2023-09-18:
Hi Dr.Jiao

I use an openwifi(antsdr) to perform as an AP, an openwifi(antsdr) to perform as a client. And I use simple socket program running on two boards to communicate with each other by tcp packets. I want to capture the CSI information of tcp packets which are transmitted by two openwifi. 

I have found that the FC field of tcp packet is 0x8802. 
[image]

So I run the code below:
```
./side_ch_ctl wh1h1001
./side_ch_ctl wh5h8802
./side_ch_ctl g
```

But the boards output like
```
loop 64 side info count 0
loop 128 side info count 0
...
```
I think it means that I can't get any CSI information with this condition.

Could you please help me?

JiaoXianjun, 2023-09-18:
If you do not set the condition, can you capture CSI?
Could you try the 0x8802 with inverted bit order, or byte order?

mkk0719, 2023-09-21:
I have succeeded in capturing the specific CSI by inverting the byte order. Thank you!"

closed 2023-09-26 by JiaoXianjun