open-sdr/openwifi #346: Support Neptune SDR B210

lejam1798, 2023-09-27:
0. Could you send email to xianjun.jiao@ugent.be to introduce your self?
_title "neptunesdr rev3 support" dated September 25, 2023_
1. Our image is used directly or you build your own image?
_openwifi-1.4.0-notter.img.xz and built from master and neptunesdr branches._
2. What is your own modification?
_---_
3. Versions: OS, Vivado, openwifi/openwifi-hw repo branch and commit revision
20.04.1-Ubuntu, Vivaldo 2021.1_0610_2318, current actual commits
4. Board/hardware type
_Neptune SDR B210, zynq7020+ad9361_
5. WiFi channel number
_---_
6. Steps to reproduce the issue, and the related error message, screenshot, etc
same as [Issue#318](https://github.com/open-sdr/openwifi/issues/318)
7. Describe your debug efforts by Linux native tools, such as tcpdump and "cat /proc/interrupts"
_---_
8. Describe your debug efforts by: https://github.com/open-sdr/openwifi/blob/master/doc/README.md#Debug-methods
_---_
9. Any other thing we need to know for helping you better? 
---
I've tried to set up an OpenWiFi on my board, but experienced the same problem as described in [issue318](https://github.com/open-sdr/openwifi/issues/318). I suspect there is a new, third revision of this board and this is why I think so:
1) At ebay and aliexpress this board named as NeptuneSDR B210 Plus (example: https://www.ebay.com/itm/354713497672),
2) There was an SD-card together with the board. This SD-card contains 2 directories in the BOOT partition named "NeptuneSDR P210" and "NeptuneSDR K210" which have the same uImage, devicetree.dtb, bootgen_sysfiles.tgz (contains fsbl.elf, system_top.bit, system_top.hdf, u-boot-zed.elf, zynq.bif) and uEnv.txt files but different BOOT.BIN files,
3) My neptune board looks just the same as in issue318, but its silkscreen name a bit differ - the letter 'p' in "neptune SDR" is aligned with the other letters (it misaligned even on the screenshots on ebay I provided above)

My current status with this board is:
1) device successfully starts with files from NeptuneSDR P210, but spi devices located not under the /sys/bus/spi/drivers/ad9361/, but under /sys/bus/spi/devices/ (contains spi0.0 spi1.0 spi1.1 spi2.0). Linux Kernel is 4.14.0,
2) device successfully starts with files from NeptuneSDR K210 and spi0.0 is located right where it should be ( /sys/bus/spi/drivers/ad9361). Linux Kernel is 4.14.0,
3) latest openwifi and openwifi-hw-img both at master branch: device successfully starts, but there is no ad9361 under the /sys/bus/spi/drivers/ (only /sys/bus/spi/drivers/spi-nor/spi1.0 is presented). Linux Kernel is 5.10.0,
4) openwifi and openwifi-hw-img both at neptunesdr branch: unable to connect to the device via ssh so there is no information about exact problem. Kernel is 5.10.0.
5) I contacted the seller to get related documentation and sw

Thanks in advance!

JiaoXianjun, 2023-11-03:
We also do not have the latest support code from the author of neptunesdr.

closed 2023-11-03 by JiaoXianjun

ltang-xt, 2024-06-14:
is this issue fixed ? and how to fixed ?