open-sdr/openwifi #357: SDR0 not loaded on ANTSDR E310

remydemy, 2023-10-30:
Hei,
I'm trying to get openwifi working on my Antsdr E310. But the `wgd.sh` script fails and therefore the `fosdem.sh` script also fails.
I flashed the `openwifi-1.4.0-notter.img`onto a 32GB SD card using Ubuntu and put the three files that are in the antsdr into the base folder of BOOT. I also checked for the `kernel_modules` and `interfaces.new` which did not exist.
After running `raspi-config --expand-rootfs` and `./openwifi/setup_once.sh`, which ran without problems, I proceeded to execute `./wdg.sh` in the `openwifi`-Folder. 
Now here comes the problem. I get following output:
root@analog:~/openwifi# ./wgd.sh 
usage:
  Script for load (or download+load) different driver and FPGA img without rebooting
  no  argument: Load .ko driver files and FPGA img (if system_top.bit.bin exist) in current dir with test_mode=0.
  1st argument: If it is a NUMBER, it will be assigned to test_mode. Then load everything from current dir.
  1st argument: If it is a string called "remote", it will download driver/FPGA and load everything.
  - 2nd argument (if exist) is the target directory name for downloading and reloading
  - 3rd argument (if exist) is the value for test_mode
  1st argument: neither NUMBER nor "remote" nor a .tar.gz file, it is regarded as a directory and load everything from it.
  - 2nd argument (if exist) is the value for test_mode
  1st argument: a .tar.gz file, it will be unpacked then load from that unpacked directory
  - 2nd argument (if exist) is the value for test_mode
 
Module                  Size  Used by
mac80211              565248  0
libarc4                16384  1 mac80211
xilinx_dma             28672  0
ad9361_drv            163840  2
cfg80211              319488  1 mac80211
8021q                  24576  0
mwipcore               16384  0
mwipcore_iio_streaming    16384  1 mwipcore
mwipcore_dma_streaming    20480  1 mwipcore
mathworks_ip_common    16384  1 mwipcore
ipv6                  397312  28
TARGET_DIR ./
DOWNLOAD_FLAG 0
test_mode 0
tx_offset_tuning_enable 0
$TARGET_DIR is found!
 
hostapd: no process found
dhcpd: no process found
sdr0: ERROR while getting interface flags: No such device
rmmod: ERROR: Module sdr is not currently loaded
./load_fpga_img.sh: line 14: warning: command substitution: ignored null byte in input
xlnx,zynq-7000
.//system_top.bit.bin
system_top.bit.bin
+ ifconfig sdr0 down
sdr0: ERROR while getting interface flags: No such device
+ rmmod sdr
rmmod: ERROR: Module sdr is not currently loaded
+ rmmod openofdm_rx
rmmod: ERROR: Module openofdm_rx is not currently loaded
+ rmmod openofdm_tx
rmmod: ERROR: Module openofdm_tx is not currently loaded
+ rmmod rx_intf
rmmod: ERROR: Module rx_intf is not currently loaded
+ rmmod tx_intf
rmmod: ERROR: Module tx_intf is not currently loaded
+ rmmod xpu
rmmod: ERROR: Module xpu is not currently loaded
+ sleep 1
+ '[' -f .//system_top.bit.bin ']'
+ echo 0
+ mkdir -p /lib/firmware
+ cp .//system_top.bit.bin /lib/firmware/ -rf
+ echo system_top.bit.bin
+ '[' xlnx,zynq-7000 '!=' xlnx,zynq-7000 ']'
+ SPI_DEVNAME=spi0.0
+ DDS_DEVNAME=79024000.cf-ad9361-dds-core-lpc
+ ADC_DEVNAME=79020000.cf-ad9361-lpc
+ '[' '!' -d /sys/bus/spi/drivers/ad9361/spi0.0 ']'
+ cd /sys/bus/spi/drivers/ad9361/
+ echo spi0.0
+ echo spi0.0
+ '[' '!' -d /sys/bus/platform/drivers/cf_axi_adc/79020000.cf-ad9361-lpc ']'
+ cd /sys/bus/platform/drivers/cf_axi_adc/
+ echo 79020000.cf-ad9361-lpc
+ echo 79020000.cf-ad9361-lpc
+ set +x
tx_offset_tuning_enable 0
openwifi_ad9361_fir_tx_0MHz_11n.ftr tx_fir_enable 1
Found openwifi_ad9361_fir_tx_0MHz_11n.ftr
+ test -f /sys/bus/iio/devices/iio:device0/in_voltage_rf_bandwidth
+ test -f /sys/bus/iio/devices/iio:device1/in_voltage_rf_bandwidth
+ cd /sys/bus/iio/devices/iio:device1/
+ set +x
FIR Rx: 48,1 Tx: 48,1
1
1
rx0 agc fast_attack
slow_attack
fast_attack
rx1 agc fast_attack
slow_attack
fast_attack
40000000
25215513
40000000
25215414
rssi
104.50 dB
110.75 dB
rx0 gain to 70
73.000000 dB
./rf_init_11n.sh: 101: echo: echo: I/O error
73.000000 dB
rx1 gain to 70
73.000000 dB
./rf_init_11n.sh: 106: echo: echo: I/O error
73.000000 dB
tx0 gain -89dB
-10.000000 dB
-89.000000 dB
tx1 gain 0dB
-10.000000 dB
0.000000 dB
rmmod: ERROR: Module tx_intf is not currently loaded
+ insmod .//tx_intf.ko
tx_intf is loaded!
rmmod: ERROR: Module rx_intf is not currently loaded
+ insmod .//rx_intf.ko
rx_intf is loaded!
rmmod: ERROR: Module openofdm_tx is not currently loaded
+ insmod .//openofdm_tx.ko
openofdm_tx is loaded!
rmmod: ERROR: Module openofdm_rx is not currently loaded
+ insmod .//openofdm_rx.ko
openofdm_rx is loaded!
rmmod: ERROR: Module xpu is not currently loaded
+ insmod .//xpu.ko
xpu is loaded!
rmmod: ERROR: Module sdr is not currently loaded
+ insmod .//sdr.ko test_mode=0
sdr is loaded!
the end

It says at the end that sdr **is loaded**. But obviously it is not because a couple lines above it's not correctly loaded.

When I then run `./fosdem.sh` I get:
root@analog:~/openwifi# ./fosdem.sh
hostapd: no process found
rm: cannot remove '/var/run/dhcpd.pid': No such file or directory
Job for isc-dhcp-server.service failed because the control process exited with error code.
See "systemctl status isc-dhcp-server.service" and "journalctl -xe" for details.
Configuration file: hostapd-openwifi.conf
sdr0: interface state UNINITIALIZED->COUNTRY_UPDATE
Using interface sdr0 with hwaddr 66:55:44:33:22:61 and ssid "openwifi"
sdr0: interface state COUNTRY_UPDATE->ENABLED
sdr0: AP-ENABLED 

Running `systemctl status isc-dhcp-server.service` results in:
root@analog:~/openwifi# systemctl status isc-dhcp-server.service
● isc-dhcp-server.service - LSB: DHCP server
     Loaded: loaded (/etc/init.d/isc-dhcp-server; generated)
     Active: failed (Result: exit-code) since Thu 2023-02-09 10:58:20 GMT; 44s ago
       Docs: man:systemd-sysv-generator(8)
    Process: 13048 ExecStart=/etc/init.d/isc-dhcp-server start (code=exited, status=1/FAILURE)
        CPU: 349ms
     CGroup: /system.slice/isc-dhcp-server.service
             └─13066 /usr/sbin/dhcpd -4 -q -cf /etc/dhcp/dhcpd.conf

What can I do, so that the sdr is loaded correctly and the dhcp server runs?

On a side note: While running the `setup_once.sh` the output said `make: warning:  Clock skew detected.  Your build may be incomplete.`, but I'm not sure if that is relevant.

Thanks for any help!
Best regards,
Emily

JiaoXianjun, 2023-10-30:
I don't see any issue with your log.
Can you connect the SSID openwifi from your phone/laptop?

remydemy, 2023-10-30:
Yes ok nvm. I was so confused by the output saying sdr0 is not loaded, that I did not try if I can connect to the ssid. Which I now tried and I can. 
Sorry for the bother nad thanks for the quick reply!

Btw whats the reason for it saying that sdr0 is not loaded correctly?

closed 2023-11-02 by JiaoXianjun